innoveit/react-native-ble-manager #239: illegal callback error

Tekumalla, 2017-11-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v^5.0.3
- react-native v0.44.0
-Android v6.0
- iOS 10+

### Expected behaviour

We should be able to connect to multiple headsets without callback errors.

### Actual behaviour

In our application we connected with one headset and given it a name. Later when connecting to another headset we got the callback error. In this process we were resetting the stack navigator and the componentWill mount function has the code that connects to the device to read the data. 
[image]
[image]
[image]```

marcosinigaglia, 2017-11-07:
Hi, I think you are calling the connection method multiple time without waiting the response.

Tekumalla, 2017-11-07:
Hello,

Is it possible to fix it from the BLE Manager library? I see a issue with the react-native navigation which is trying to bring up all the views when it is reset.
Due to the dispatch reset event, all views componentWillMount() function of the navigation stack views is getting triggered. We have the device connection in the componentwillmount() which is leading to multiple connections.

With Regards,
Sai

marcosinigaglia, 2017-11-07:
No, you have to handle this things in your app.

closed 2018-01-11 by marcosinigaglia