innoveit/react-native-ble-manager #241: how to know if a device is supported

pstanton, 2017-11-09:
Sorry, this may not be a bug ...
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.50.3
- iOS/Android v7.0

### Expected behaviour

Running the example app, I have 3 bluetooth devices nearby and broadcasting - a nobsound audio device (Bluetooth 4.0 apparently), Lenovo thinkpad and Samsung galaxy S5. I would expect one of these devices to appear in the "peripherals"?

### Actual behaviour

App.js:58  Permission is OK
App.js:112 Scanning...
App.js:105 Scan is stopped
App.js:74 App has come to the foreground!
App.js:76 Connected peripherals: 0

pstanton, 2017-11-12:
I tried the demo app again in another location, this time with many blue-tooth devices and it only worked with one - a device that didn't publish a name so i was unable to figure out what it was. devices that didn't work included some 4.1 BLE devices that i could pair and connect to via the native bluetooth manager, and connect to thru different apps - so what determines whether "scan" picks up devices? is this project flawed?

marcosinigaglia, 2017-11-13:
Hi, I think that depends on the device configuration, you can use this project to test https://github.com/sandeepmistry/bleno/tree/master/examples/pizza .

pstanton, 2017-11-13:
hi marcos, thanks for the tip. the pizza demo looks very similar to the demo [here](https://github.com/innoveit/react-native-ble-manager/tree/master/example) - or do you think it does anything more to support peripherals?

can you offer any concrete info regarding peripheral configuration i can chase up? we are going to need to integrate with a particular chip but it might be possible to configure this chip/driver accordingly.

my concern is that this chip does seem to work with other android apps, so the failure point seems to be this project (or how i have my implementation/app set up).

marcosinigaglia, 2017-11-14:
Hi, you can use other app, like BLE scanner, to check the chip.

closed 2018-01-11 by marcosinigaglia