innoveit/react-native-ble-manager #243: Possible Unhandled Promise Rejection (id: 0): "Peripheral not found or not connected"

kf729ik, 2017-11-16:
### Version

Tell us which versions you are using:

- react-native-ble-manager: "react-native-ble-manager": "^6.1.1",
- "react-native": "0.48.3",
- iOS/
Hi all,
### Expected behaviour

I would like to scan, connect, disconnect to a bluetooth device for x times.

### Actual behaviour

Sometime if I repeat the above mentioned process it crash with: 
Possible Unhandled Promise Rejection (id: 0): "Peripheral not found or not connected"

### Steps to reproduce

1. Scan for device and connect, rerouting to an another view
2. Receiving datas, process them; disconnect from the device after specific time e.g after 10s of live data OR disconnecting by pressing the back navigation button
3. After I pressed the back button (it reroutes me to the previous one, where I can connect to the device)
4. Pressing scan, connect
5. Sometimes it works 4,5,6 times without problem, sometimes it crash on the 2nd attempt.

### Stack trace and console log
In my log everything seems to be ok, every time when I scan it's searching for the devices > I get a list of Devices (from those I pick one specific what I need), if I press the connect button the log updates that I connected or disconnected (if I press disconnect). 

This process repeat all the time, also if the app crash, but in that case after connecting to the device I got that message in yellow messagebox in the bottom of the phone screen. After that it's dead, it's not opening to new screen and also do not receive any information from the device.

Thanks for any help,
I really appreciate it :)

robseaman, 2017-12-13:
I'm seeing the same issue with IOS with react-native 51 and react-native-ble-manager 6.2.4. My failure rate is higher on reconnects but the sequence is the same.

kf729ik, 2017-12-13:
@robseaman  try to call the BleManager.start() method only once, if you place it to the BleManager class it will call every time when this class is called and that's why it crashes in the app.

robseaman, 2017-12-13:
Thank you @kf729ik for the response. That is exactly what I was doing, calling BleManager.start() each time my component was mounted.

closed 2017-12-13 by marcosinigaglia

Poukai, 2018-07-17:
@robseaman Do you have a sample code that i can look at ?

soadtech, 2021-09-07:
I have the same problem, I see that they only close the issues but there is no response