innoveit/react-native-ble-manager #245: Unpair function?

msheldon, 2017-11-20:
Hi, does your library has unpair function? because I saw that there is a createBond method but I still not see any docs with regards to unpairing bonded peripherals.

marcosinigaglia, 2017-11-20:
Hi, there is no method, you can implement and make a PR

marcosinigaglia, 2018-01-11:
Hi @msheldon , can you try the removeBond method in the latest commit?

marcosinigaglia, 2018-05-08:
Hi @yorikcv , thanks, try the new release. (6.3.0)

yorikcv, 2018-05-10:
@marcosinigaglia Thanks!

marcosinigaglia, 2018-05-10:
@yorikcv tell me if it's working that I add the method in the documentation.

yorikcv, 2018-05-10:
@marcosinigaglia I have tested it on Nexus 6P, Nexus 5 and OnePlus 3. In 3 phones it works.

marcosinigaglia, 2018-05-10:
@yorikcv perfect.

closed 2018-05-10 by marcosinigaglia

ducNgbk, 2018-05-29:
Hello,
I am doing a very simple function.
What I expect here is when I run below code, I'll get `removeBond success`, and my device will be deleted from Bluetooth setting of my phone.
However, I got `removeBond success`, and my device was still there in Bluetooth Setting
I've tested on HTC desire 826 Android 6.0, Vivo 1610 Android 6.0.1

My dependencies:
```
    "react": "^16.3.0-alpha.1",
    "react-native": "0.54.3",
    "react-native-ble-manager": "^6.2.8",
    "react-native-elements": "^0.19.1",
    "react-native-fbsdk": "^0.7.0",
    "react-native-firebase": "^4.2.0",
    "react-native-google-sign-in": "^1.0.16",
    "react-navigation": "^1.5.2",
    "react-redux": "^5.0.7",
    "redux": "^3.7.2"
```
My code:
```
      console.log("Delete bond by code");
      await BleManager.removeBond(deviceInfo.addr)
        .then(() => {
          console.log("removeBond success");
        })
        .catch(() => {
          console.log("fail to remove the bond");
        });
```