innoveit/react-native-ble-manager #247: Notification doesn't trigger BleManagerDidUpdateValueForCharacteristic Event

elailai94, 2017-11-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.50.3
- Android v7

### Expected behaviour

When starting notification, we should receive data sent by BLE device, which causes the BleManagerDidUpdateValueForCharacteristic event to trigger.

### Actual behaviour

It starts notification, but nothing comes in from BLE device, but the BleManagerDidUpdateValueForCharacteristic event is not triggered.

### Steps to reproduce

1. Connect to device
2. Start notification
3. See if it receives any data

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2017-11-29:
Hi, it depends how the ble device is working. Try with an other app like BLE Scanner if it's working.

elailai94, 2017-11-30:
Hi, thank you for your quick reply. I managed resolved this issue after looking into it. This issue is actually related to the fact that I modified the library to request a new MTU size upon being connected with the BLE device. The service discovery/set characteristic notification was occurring before the MTU request was completed. Here's the Medium article which helped me: https://www.medium.com/@willweizong/2017-05-02-ble-mtu-1a223b73cb5a.

closed 2017-12-21 by elailai94