innoveit/react-native-ble-manager #248: iOS: Error Domain=CBATTErrorDomain Code=13 "The value's length is invalid." UserInfo={NSLocalizedDescription=The value's length is invalid.}

melkayam92, 2017-11-29:
Tell us which versions you are using:

- react-native-ble-manager v6.2.1
- react-native v0.42.3
- iOS/Android v - iOS 11

### Expected behaviour

Try to write to device, The iOS device should write to the ble device.

### Actual behaviour
I get an error: 

Error Domain=CBATTErrorDomain Code=13 "The value's length is invalid." UserInfo={NSLocalizedDescription=The value's length is invalid.}

****
Same writes work well on Android, and also worked well on iOS 10..

any idea?

marcosinigaglia, 2017-11-29:
Probably the value you write is too long.

melkayam92, 2017-11-29:
Hey,

I just write the value 0, so i don't think that's the case.
also the same thing worked well in iOS 10 + working now in Andorid.

rushstrike, 2018-01-03:
I am having the exact same issue it used to work on iOS 10. I am not even using react and just standard BLE core bluetooth api.

Error Domain=CBATTErrorDomain Code=13 "The value's length is invalid." UserInfo={NSLocalizedDescription=The value's length is invalid.}

IshwaryaRaja, 2018-03-23:
I have the same problem. Please suggest a fix

sarangpm18, 2019-01-29:
Hi, Anyone got the fix?

marcosinigaglia, 2019-01-29:
Hi @sarangpm18 , try to debug and solve, I never had this problem.

nehasharma1101, 2020-02-21:
I am also facing same issue. Please suggest some solution.
I think because of this issue, I am not able to transfer data to my ble device.

marcosinigaglia, 2020-02-21:
Hi @nehasharma1101 , I can't reproduce your problem.

nehasharma1101, 2020-02-21:
This is the code and im getting this error 
var data = this.convertStringToByteArray("Hello");

                  setTimeout(() => {
                    BleManager.startNotification(
                      peripheral.id,
                      peripheral.advertising.serviceUUIDs[0],
                      charactersitic[0].characteristic
                    )
                      .then(() => {
                        console.log(
                          "Started notification on " + peripheral.id,
                          peripheral.advertising.serviceUUIDs[0],
                          charactersitic[0].characteristic
                        );
                        setTimeout(() => {
                          BleManager.write(
                            peripheral.id,
                            peripheral.advertising.serviceUUIDs[0],
                            charactersitic[0].characteristic,
                            data

                            // ["01100110", "01101111", "01101111"]

                            // data1.length
                          ).then(() => {
                            console.log("Writed NORMAL crust");
                            BleManager.write(
                              peripheral.id,
                              peripheral.advertising.serviceUUIDs[0],
                              charactersitic[0].characteristic,
                              data

                              // ["01100110", "01101111", "01101111"]

                              // data1.length
                            ).then(() => {
                              console.log(
                                "Writed 351 temperature, the pizza should be BAKED"
                              );
                            });
                          });
                        }, 500);
                      })
                      .catch(error => {
                        console.log("Notification error", error);
                      });
                  }, 200);

marcosinigaglia, 2020-02-21:
Hi @nehasharma1101 , the example code is not general, you have to know in which service/characteristic to write, the format, etc.

FazilMuhammed, 2020-10-12:
@marcosinigaglia @nehasharma1101  Hello guys any possible to write image on printer Please help me..

acro5piano, 2021-08-17:
I found a solution: I must decode UTF8 string before writing out.

so, change

```typescript
const { stringToBytes } = require('convert-string');

const data = stringToBytes('myText')
await BleManager.write(connectedDeviceId, serviceUUID, characteristicUUID, data, 1_000_000);
```

to

```typescript
import utf8 from 'utf8';
const { stringToBytes } = require('convert-string');

const data = stringToBytes(utf8.encode('myText'))
await BleManager.write(connectedDeviceId, serviceUUID, characteristicUUID, data, 1_000_000);
```

closed 2023-02-10 by marcosinigaglia

SeeTheHi, 2025-03-17:
> 我找到了一个解决方案：我必须在写出之前解码 UTF8 字符串。
> 
> 因此，改变
> 
> const { stringToBytes } = require('convert-string');
> 
> const data = stringToBytes('myText')
> await BleManager.write(connectedDeviceId, serviceUUID, characteristicUUID, data, 1_000_000);
> 到
> 
> import utf8 from 'utf8';
> const { stringToBytes } = require('convert-string');
> 
> const data = stringToBytes(utf8.encode('myText'))
> await BleManager.write(connectedDeviceId, serviceUUID, characteristicUUID, data, 1_000_000);

It's not possible, my will is UTF-8, and the idol returns such an error