innoveit/react-native-ble-manager #249: connect method hands on iOS if peripheral is not available

clarifymed, 2017-11-30:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.3
- react-native v42.3
- iOS/Android iOS (9 through 11)

### Expected behaviour

The connect should reject within a reasonable period of time when the peripheral is not turned on

### Actual behaviour

connect never returns. On android it returns with a reject within a couple of seconds

### Steps to reproduce

1.  bleManager.connect(<peripheral id>)
2. 
3.

Note, this worked correctly when I used a much older version 3.x

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

garbo-a, 2018-02-01:
Confirming this bug on iOS

whitedogg13, 2018-04-12:
FYI, I just made a PR #313 to explain this is iOS's design rather than a library issue.

closed 2018-06-19 by marcosinigaglia