innoveit/react-native-ble-manager #250: Periodic disconnection from the device after startNotification and BleManagerDidUpdateValueForCharacteristic events

elailai94, 2017-11-30:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.50.3
- Android v7

### Expected behaviour

Once we are connected to the device, called startNotification and subscribed to BleManagerDidUpdateValueForCharacteristic events, we should be constantly connected to the device and receiving characteristic update notifications unless we manually disconnect from the device or the device is out of range.

### Actual behaviour

Once we are connected to the device, called startNotification and subcribed to BleManagerDidUpdateValueForCharacteristic events, we are able to read the value from the BleManagerDidUpdateValueForCharacteristic events. After some time, the app will automatically disconnect itself from the device. There is usually a warning that comes before, which states 'Possible Unhandled Promise Rejection (id: 0): "Set notification failed for 0000fff1-0000-1000-8000-00805f9b34fb".

### Steps to reproduce

1. Check the device to make sure it is updating the characteristic value
2. connect(...)
3. retrieveServices(...)
4. startNotification(...)
5. Read value from BleManagerDidUpdateValueForCharacteristic events

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

[image]
[image]

marcosinigaglia, 2017-12-04:
Hi, maybe is the device that disconnect you?

elailai94, 2017-12-07:
It could well be the device is disconnecting from the app. I will check it on the device side. Other than the example in the repository, do you have a more full-featured, complete example of how to use this library? Maybe the code that you test this library with?

closed 2017-12-21 by elailai94