innoveit/react-native-ble-manager #251: Support BLE in the browser

circuitry2, 2017-12-01:
It would be great to have BLE support in the browser as well, using Bluetooth Web API

marcosinigaglia, 2017-12-06:
If you use the Bluetooth Web API you don't need a module.

circuitry2, 2017-12-06:
@marcosinigaglia the Bluetooth Web API is not supported in webviews, so this will not work without a module.

marcosinigaglia, 2017-12-06:
I don't get you point, why is better to use a fake Bluetooth Web API instead the module directly?

marcosinigaglia, 2018-01-02:
Hi @circuitry2 , only chrome support BLE on the browser in experimental way so i don't get you point.

closed 2018-01-11 by marcosinigaglia

IjzerenHein, 2018-08-16:
It would be useful though to have the web-bluetooth API available on react-native, so we can write cross platform code that will work on the browser as well