innoveit/react-native-ble-manager #252: Possible GATT disconnection issue on Samsung devices

eltodesc, 2017-12-01:
Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.44.0
- Android S7(SM-G930U) 6.0.1,  Samsung J7 7.0  &  Samsung A5 7.0

I have an app that scans for particular types of BLE devices, connects to each serially, enables notifications, processes said notifications, disables notifications (once all are received), disconnects and removes the peripheral from the cache (whatever this does...).  This appears to work well on iOS and non-Samsung Android devices.  However on Samsung devices if I perform this operation more than 10 times in a row, I get into a state where I can no longer scan for BLE devices and the BleManagerStopScan callback is immediately fired after a startScan attempt.  The only way to resolve this issue is to restart the app.  Please note that I have encapsulated the BLEManager into a singleton object and BLEManager.start is only being called once per launch of the app.   Looking at the log I see:

12-01 11:55:09.881 14776-14822/com.xxxx.xxxx D/logs: scan
12-01 11:55:09.891 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.891 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.891 14776-14822/com.xxxx.xxxx D/BluetoothLeScanner: Start Scan
12-01 11:55:09.891 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.901 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.901 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.901 14776-14822/com.xxxx.xxxx D/BluetoothAdapter: STATE_ON
12-01 11:55:09.951 14776-14788/com.xxxx.xxxx D/BluetoothLeScanner: onClientRegistered() - status=133 clientIf=0
12-01 11:55:09.951 14776-14788/com.xxxx.xxxx D/BluetoothLeScanner: Registration failed, unregister clientIf = 0
12-01 11:55:09.951 14776-14822/com.xxxx.xxxx D/BluetoothLeScanner: Scan failed, reason app registration failed for UUID = 9c5c1374-6492-4f13-88d9-671fb60df0ca

Browsing the web I see that this registration failure could be caused by too many open gatt connections, however, I am explicitly closing the connection and can verify that the BLE peripheral is in fact getting disconnected successfully.  Another thing worth noting is that upon peripheral disconnection I see the following exception:

12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/logs: Disconnect from: C5:12:11:A4:95:02
12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/BluetoothGatt: cancelOpen() - device: C5:12:11:A4:95:02
12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/BluetoothGatt: close()
12-01 11:22:29.881 14776-14787/com.xxxx.xxxx D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=10 device=C5:12:11:A4:95:02
12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/BluetoothGatt: unregisterApp() - mClientIf=10
12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/logs: Disconnect
12-01 11:22:29.881 14776-14822/com.xxxx.xxxx D/logs: Peripheral event (BleManagerDisconnectPeripheral):C5:12:11:A4:95:02
12-01 11:22:29.881 14776-14787/com.xxxx.xxxx W/BluetoothGatt: Unhandled exception in callback
                                                                       java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattCallback.onConnectionStateChange(android.bluetooth.BluetoothGatt, int, int)' on a null object reference
                                                                           at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
                                                                           at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
                                                                           at android.os.Binder.execTransact(Binder.java:453)
12-01 11:22:29.891 14776-14822/com.xxxx.xxxx D/logs: Removing from list: C5:12:11:A4:95:02
12-01 11:22:29.891 14776-14821/com.xxxx.xxxx I/ReactNativeJS: BLE_LOG: Disconnected from 5B95

This exception does not seem to cause an immediate issues in my app but I thought it worth noting.   Once more thing that is driving me crazy on my Samsung device is that when I call BleManager.connect, the BleManagerDisconnectPeripheral callback is getting fired exactly 5 seconds after this call if the advertisment packet is not seen/heard.  It should be noted that my BLE peripheral has an advertising interval of 15 seconds, so I have built in retry logic to account for this strange 5 second connection timeout.  This is not ideal to say the least as I am sometime missing the advertisement packet during this retry attempt an have to wait another 15s.  This does not happen on iOS.   Is this connection timeout configurable?

Please help!!  Thank you so much

marcosinigaglia, 2017-12-04:
Hi, unfortunately the android's bluetooth works differently in different os version or device.

eltodesc, 2017-12-04:
Is anyone else reporting such behavior?  Is there currently any way to set a connection timeout or is this a feature that is in the backlog?

garbo-a, 2018-02-01:
Having the same issue with Android phones (versions 6, 7 and 8), especially Samsung phones
On attempt to connect to device getting STATUS = 133 (gatt error) and event BleManagerDisconnectPeripheral
`D/logs: Get connected peripherals
D/logs: Connect to: 00:14:EE:0C:6B:2E
D/libEGL: eglInitialize EGLDisplay = 0x98f9a614
D/BluetoothGatt: connect() - device: 00:14:EE:0C:6B:2E, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=b93a51b4-f8f9-403e-b355-7e4954f35add
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
D/libEGL: eglInitialize EGLDisplay = 0x98f9a614
D/libEGL: eglInitialize EGLDisplay = 0x98f9a614
D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=6 device=00:14:EE:0C:6B:2E
D/logs: onConnectionStateChange from 133 to 0 on peripheral:00:14:EE:0C:6B:2E
D/logs: Peripheral event (BleManagerDisconnectPeripheral):00:14:EE:0C:6B:2E`
Still looking for fix..

sowdri, 2018-03-26:
I can confirm that this happens on Nexus 5 (LG) device as well. Works well with IOS but cannot establish connection on Android phones.
```
03-26 18:29:50.109  5259  9201 D logs    : Get connected peripherals
03-26 18:29:50.109  5259  9201 D BluetoothManager: getConnectedDevices
03-26 18:29:50.119  5259  9201 D logs    : Request bond to: DC:0D:30:06:91:20
03-26 18:29:50.146  5259  9201 D logs    : Connect to: DC:0D:30:06:91:20
03-26 18:29:50.147  5259  9201 D BluetoothGatt: connect() - device: DC:0D:30:06:91:20, auto: false
03-26 18:29:50.147  5259  9201 D BluetoothGatt: registerApp()
03-26 18:29:50.148  5259  9201 D BluetoothGatt: registerApp() - UUID=540aeed2-e78e-479f-aa12-ded53729e1b8
03-26 18:29:50.151  2740  3249 I bt_stack: [INFO:gatt_api.cc(1004)] GATT_Register
03-26 18:29:50.151  2740  3249 I bt_stack: [INFO:gatt_api.cc(1027)] allocated gatt_if=12
03-26 18:29:50.152  5259  5273 D BluetoothGatt: onClientRegistered() - status=0 clientIf=12
03-26 18:29:50.154  2740  2835 D bt_btif_config: btif_get_address_type: Device [dc:0d:30:06:91:20] address type 0
03-26 18:29:50.154  2740  2835 D bt_btif_config: btif_get_device_type: Device [dc:0d:30:06:91:20] type 3
03-26 18:29:50.154  2740  3249 I bt_stack: [INFO:gatt_api.cc(1168)] GATT_Connectgatt_if=12 dc:0d:30:06:91:20
03-26 18:29:50.621   767   782 I UsageStatsService: User[0] Flushing usage stats to disk
03-26 18:29:51.779  2740  3249 W bt_btif : bta_dm_acl_change info: 0x10
03-26 18:29:51.779  2740  2835 D bt_btif_dm: remote version info [dc:0d:30:06:91:20]: 0, 0, 0
03-26 18:29:51.824  3503  3503 W SearchService: Abort, client detached.
03-26 18:29:51.831  3503  3503 E DeviceDetector: UUID = 00001101-0000-1000-8000-00805f9b34fb
03-26 18:29:51.831  3503  3503 E DeviceDetector: UUID = 00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:51.831  3503  3503 E DeviceDetector: UUID = 00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:51.835  2740  3249 I bt_bta_dm: bta_dm_find_services services_to_search=7fffffff
03-26 18:29:51.835  2740  3249 I bt_bta_dm: bta_dm_find_services search UUID = 1200
03-26 18:29:52.292  2740  3249 W bt_l2cap: L2CAP - no CCB for conn rsp, LCID: 0 RCID: 0
03-26 18:29:52.313  2740  3249 W bt_sdp  : process_service_search_attr_rsp
03-26 18:29:52.314  2740  3249 W bt_sdp  : sdp_copy_raw_data: list_len:0 cpy_len:4096 p:0x7c4799d003 p_ccb:0x7c3c4ead80 p_db:0x7c3a08b000 raw_size:4096 raw_used:0 raw_data:0x7c3c3fd528
03-26 18:29:52.321  2740  3249 I bt_bta_dm: bta_dm_find_services services_to_search=7ffffffe
03-26 18:29:52.322  2740  3249 I bt_bta_dm: bta_dm_find_services search UUID = 0100
03-26 18:29:52.364  2740  3249 W bt_sdp  : process_service_search_attr_rsp
03-26 18:29:52.364  2740  3249 W bt_sdp  : sdp_copy_raw_data: list_len:67 cpy_len:67 p:0x7c4799d003 p_ccb:0x7c3c4ead80 p_db:0x7c3a08b000 raw_size:4096 raw_used:0 raw_data:0x7c3c3fd528
03-26 18:29:52.371  2740  2835 I bt_btif_dm: btif_dm_search_services_evt index:0 uuid:00001101-0000-1000-8000-00805f9b34fb
03-26 18:29:52.371  2740  2835 I bt_btif_dm: btif_dm_search_services_evt index:1 uuid:00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:52.371  2740  2835 I bt_btif_dm: btif_dm_search_services_evt index:2 uuid:00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:52.397  3351  3351 I GsaVoiceInteractionSrv: O received Intent { act=android.bluetooth.device.action.UUID flg=0x10 (has extras) }
03-26 18:29:52.400  2740  2740 D BluetoothPhonePolicy: Received ACTION_UUID for device DC:0D:30:06:91:20
03-26 18:29:52.400  2740  2740 D BluetoothPhonePolicy: index=0uuid=00001101-0000-1000-8000-00805f9b34fb
03-26 18:29:52.400  2740  2740 D BluetoothPhonePolicy: index=1uuid=00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:52.401  2740  2740 D BluetoothPhonePolicy: index=2uuid=00000000-0000-1000-8000-00805f9b34fb
03-26 18:29:52.401  2740  2740 D BluetoothPhonePolicy: processInitProfilePriorities() - device DC:0D:30:06:91:20
03-26 18:29:52.401  2740  2740 D HidService: getHidService(): returning com.android.bluetooth.hid.HidService@7779615
03-26 18:29:52.409  2767  2891 D LocalBluetoothProfileManager: Current Profiles[]
03-26 18:29:52.409  2767  2891 D BluetoothMap: getConnectedDevices()
03-26 18:29:52.416  7814  7814 D LocalBluetoothProfileManager: Current Profiles[]
03-26 18:29:52.416  7814  7814 D BluetoothMap: getConnectedDevices()
03-26 18:29:52.423  7814  7814 D BluetoothMap: getConnectionState(DC:0D:30:06:91:20)
03-26 18:29:52.425  7814  7814 D MapProfile: getConnectionStatus: status is: 0
03-26 18:29:52.425  7814  7814 D LocalBluetoothProfileManager: New Profiles[]
03-26 18:29:52.427  2767  2891 D BluetoothMap: getConnectionState(DC:0D:30:06:91:20)
03-26 18:29:52.429  2767  2891 D MapProfile: getConnectionStatus: status is: 0
03-26 18:29:52.429  2767  2891 D LocalBluetoothProfileManager: New Profiles[]
03-26 18:29:52.445  2740  2745 I zygote64: Do partial code cache collection, code=27KB, data=21KB
03-26 18:29:52.445  2740  2745 I zygote64: After code cache collection, code=27KB, data=21KB
03-26 18:29:52.445  2740  2745 I zygote64: Increasing code cache capacity to 128KB
03-26 18:29:52.851   767  6213 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.bluetooth.device.action.UUID flg=0x10 (has extras) } to com.google.android.googlequicksearchbox/com.google.android.apps.gsa.broadcastreceiver.external.ExternalCommonBroadcastReceiver
03-26 18:29:52.869  3503  3503 W SearchService: Abort, client detached.
[... cut ...]
03-26 18:30:34.214  2740  2740 D BluetoothMapService: onReceive
03-26 18:30:34.214  2740  2740 D BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
03-26 18:30:34.214  2740  2740 D BluetoothPbapService: action: android.bluetooth.device.action.ACL_DISCONNECTED
03-26 18:30:34.215  2740  2740 D BluetoothPbapService: state: -2147483648
03-26 18:30:34.215  2740  2740 W BluetoothPbapService: Unrecognized intent!
03-26 18:30:34.229  3503  3503 W SearchService: Abort, client detached.
```

Hope this helps in debugging.

closed 2018-06-14 by marcosinigaglia

anwarhamr, 2020-03-12:
I am experiencing something similar I believe on an Android 6 medical grade device made by advantech.  I connect to 4, bonded devices, I have a 2 notification characteristics setup one of which gets data every ~6 seconds.  This all works fine for a while, like hours, then slowly devices disconnect and scanning for devices that are advertising do not get picked up.  
I see a couple things in the logs that are pointing to running out of BluetoothGATT instances, as described [here](https://stackoverflow.com/questions/43962696/register-with-gatt-stack-failed-error-ble-android-samsung).  There are only 32 BluetoothGATT objects available for the system from what I'm reading/understanding.  Something seems to be holding onto these or something else that I can't fathom.  The code does not appear to be doing this in RN-ble-mgr but I'm not a Java programmer; perhaps the library we're using?  I have a counter for when devices connect/disconnect and it's not happening so it may not have anything to do with those two actions. 

I see these messages
```
03-12 09:04:53.515 13449 13484 D BluetoothAdapter: STATE_ON
03-12 09:04:53.518  7079  7092 D BtGatt.GattService: registerClient() - UUID=28ff312c-ec14-4f97-a90f-eb46fdb34265
03-12 09:04:53.518  7079  7106 E bt_btif : Register with GATT stack failed.
03-12 09:04:53.518  7079  7106 E bt_btif : Register with GATT stack failed.
03-12 09:04:53.518  7079  7098 D BtGatt.GattService: onClientRegistered() - UUID=28ff312c-ec14-4f97-a90f-eb46fdb34265, clientIf=0
03-12 09:04:53.518 13449 13463 D BluetoothLeScanner: onClientRegistered() - status=133 clientIf=0
```

and 

```
03-12 05:00:36.545  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=3 connected=0 conn_id=259 reason=0x0008
03-12 05:00:36.546  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=4 connected=0 conn_id=260 reason=0x0008
03-12 05:00:36.546  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0008
03-12 05:00:36.546  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=6 connected=0 conn_id=262 reason=0x0008
03-12 05:00:36.546  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=7 connected=0 conn_id=263 reason=0x0008
03-12 05:00:36.546  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=8 connected=0 conn_id=264 reason=0x0008
03-12 05:00:36.547  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=9 connected=0 conn_id=265 reason=0x0008
03-12 05:00:36.547  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=10 connected=0 conn_id=266 reason=0x0008
03-12 05:00:36.547  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=11 connected=0 conn_id=267 reason=0x0008
03-12 05:00:36.547  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=12 connected=0 conn_id=268 reason=0x0008
03-12 05:00:36.548  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=13 connected=0 conn_id=269 reason=0x0008
03-12 05:00:36.548  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=14 connected=0 conn_id=270 reason=0x0008
03-12 05:00:36.548  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=15 connected=0 conn_id=271 reason=0x0008
03-12 05:00:36.548  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=16 connected=0 conn_id=272 reason=0x0008
03-12 05:00:36.549  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=17 connected=0 conn_id=273 reason=0x0008
03-12 05:00:36.549  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=18 connected=0 conn_id=274 reason=0x0008
03-12 05:00:36.549  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=19 connected=0 conn_id=275 reason=0x0008
03-12 05:00:36.549  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=20 connected=0 conn_id=276 reason=0x0008
03-12 05:00:36.549  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=21 connected=0 conn_id=277 reason=0x0008
03-12 05:00:36.550  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=22 connected=0 conn_id=278 reason=0x0008
03-12 05:00:36.550  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=23 connected=0 conn_id=279 reason=0x0008
03-12 05:00:36.550  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=24 connected=0 conn_id=280 reason=0x0008
03-12 05:00:36.550  7079  7106 W bt_btif : bta_gattc_conn_cback() - cif=25 connected=0 conn_id=281 reason=0x0008
03-12 05:00:36.551  7079  7106 I bt_btm_sec: btm_sec_disconnected clearing pending flag handle:65 reason:8
03-12 05:00:36.553  7079  7098 D BtGatt.GattService: onDisconnected() - clientIf=25, connId=281, address=80:EA:CA:70:01:07
```