innoveit/react-native-ble-manager #253: Only last callback is called when there're multiple writes

allenhsu, 2017-12-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.50.1
- iOS/Android iOS 11

### Expected behaviour

When there're multiple writes, there should be multiple corresponding callbacks.

### Actual behaviour

If multiple writes are writing to the same peripheral and same characteristic, the later callback will overwrite the earlier one.

### Steps to reproduce

1. Trigger multiple writes to the same peripheral and same characteristic

### Stack trace and console log

https://github.com/innoveit/react-native-ble-manager/blob/master/ios/BleManager.m#L435

Here the key is peripheral uuid + characteristic uuid, so it's overwritten.

allenhsu, 2017-12-04:
I'm not sure if `didWriteValueForCharacteristic` is always called in the same order of `writeValue: forCharacteristic:`. Suppose the writes on native side is queued, I think the order should be same. If so, we can change the value `writeCallbacks` to array (queue) of callbacks instead of single callback so when `didWriteValueForCharacteristic`, pop and invoke the callback.

allenhsu, 2017-12-04:
Another solution or maybe better solution is queue the writes up, either on JS side or native side, so before the last write is finished, do not start processing the next write.

I'll propose a native side solution with a pr.

allenhsu, 2017-12-04:
Also, when reading values, the callback is called in:

https://github.com/innoveit/react-native-ble-manager/blob/master/ios/BleManager.m#L59

But when there's any error, the method returns, without calling and cleaning up the readCallbacks, it'll cause a pending Promise on JS side.

marcosinigaglia, 2017-12-04:
Hi, now you have to handle the queue in the js side.

allenhsu, 2017-12-05:
@marcosinigaglia do you have any best practice of queueing writes?

allenhsu, 2017-12-05:
Btw, as mentioned above, I think the readcallback should still be called when there's error occurred.

allenhsu, 2017-12-09:
However in my personal opinion, if queueing is necessary for the writes working properly, it should be in the lib, either on native side or in BleManager.js, rather than leaving it to the app.

In my use case, after connecting a device, I'll send some commands to it for data syncing, for example something like:

```
async connect(peripheral) {
  const pid = peripheral.id;
  await BleManager.connect(pid);
  const peripheralInfo = await BleManager.retrieveServices(pid);
  await BleManager.startNotification(pid, SERVICE_UUID, CHAR_UUID);
  await BleManager.write(pid, SERVICE_UUID, CHAR_UUID, command_1);
  await BleManager.write(pid, SERVICE_UUID, CHAR_UUID, command_2);
  await BleManager.write(pid, SERVICE_UUID, CHAR_UUID, command_3);
}
  ```

That's when I noticed sometimes some commands never calls back, neither `then` nor `error`. It looks weird to me because the number of callbacks is less then the number of total Promises, and it's unpredictable and unexpected.

closed 2018-01-11 by marcosinigaglia

ydykevin, 2018-09-04:
@marcosinigaglia 

Hi, I am having the problem that startNotification might get stuck on Android, maybe it is because I proceed to next operation before the callback.

Can you show me how you use redux to put the operations (like startNotification) into queue and then execute them safely (wait for the callback)?