innoveit/react-native-ble-manager #254: EnableBluetooth doesn't trigger "then"

poptocrack, 2017-12-05:
### Version

- react-native-ble-manager v6.2.3
- react-native v0.46 (expo detached)
- iOS/Android v6.0.1

### Expected behaviour

When using `BleManager.enableBluetooth()` the user accept and the `console.log()` is displayed and the bluetooth activated. The problem doesn't occur when the Bluetooth is activated before the app is opened.

### Actual behaviour

Actually, if the bluetooth is off before the app launch, the popup is shown, the user accept but the `console.log()` in the `.then()` is not displayed

### Steps to reproduce
    BleManager.enableBluetooth()
    .then(() => {
      //after user accept with the poptu, nothing is displayed in the console.
      console.log('The bluetooh is already enabled or the user confirm');
     })
     .catch((error) => {
       console.log('The user refuse to enable bluetooth');
      });

Am I not using it as it's meant?

Thanks for your job!

marcosinigaglia, 2017-12-28:
Hi, I don't think is possible to handle the situation because you are outside the app. I think you have to check the situation when the app is active again.

raguct25, 2018-07-13:
@marcosinigaglia Hi, am also facing same issue enableBluetooth() throw the only error block

raguct25, 2018-07-13:
**Version**
"react-native-ble-manager": "^6.4.0",

**code here:** 

`componentDidMount() {
    setTimeout(() => {
      SplashScreen.hide();
    }, 1000);
    BleManager.enableBluetooth()
      .then(() => {
        Alert.alert('The bluetooth is on');
      })
      .catch((error) => {
        console.log('The bluetooth is off' + error);
      });
  }`

If bluetooth on or off am getting this **console** :  The bluetooth is offTypeError: Cannot read property 'enableBluetooth' of undefined

raguct25, 2018-07-13:
Hi, It's my mistake I didn't add React Package. now I added React Package in  **android/app/src/main/java/com/project_name/MainApplication.java**

**import:**
import it.innove.BleManagerPackage;
**package:**
**getPackages() inside declare :** new BleManagerPackage()

closed 2018-07-13 by poptocrack