innoveit/react-native-ble-manager #255: why can‘t get some device name only device id

qiansitong, 2017-12-06:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2017-12-06:
I don't, are you sure that the device has a name?

todorone, 2017-12-09:
@marcosinigaglia I have same problem, on Android.
The device has the name in Android's bt manager, but on `BleManagerDiscoverPeripheral` event, object with only `advertising` , `id` and `rssi` properties is received. It's applicable not for all, but some devices...

marcosinigaglia, 2017-12-11:
Try with an another app like BLE scanner.

todorone, 2017-12-11:
@marcosinigaglia I've just tried BLE scanner, it identifies name correctly, while BLE manager returns only 3 properties without `name`...

todorone, 2017-12-11:
@marcosinigaglia Seems everything works correctly. Sorry for disturbance... :)

closed 2017-12-11 by marcosinigaglia

EduardoRicardez, 2020-05-21:
I got name as null in android always 
[Thu May 21 2020 04:48:11.500]  LOG      Got ble peripheral {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIXQeHdW0BDNFSAlCezSr5YNuatRWunTgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}, "id": "4D:07:36:9B:5A:CE", "name": null, "rssi": -65}
[Thu May 21 2020 04:48:11.860]  LOG      Got ble peripheral {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIXQeHdW0BDNFSAlCezSr5YNuatRWunTgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}, "id": "4D:07:36:9B:5A:CE", "name": null, "rssi": -65}
[Thu May 21 2020 04:48:11.185]  LOG      Got ble peripheral {"advertising": {"isConnectable": false, "manufacturerData": {"CDVType": "ArrayBuffer", "bytes": [Array], "data": "Hv8GAAEJIAIXQeHdW0BDNFSAlCezSr5YNuatRWunTgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA="}, "serviceData": {}, "serviceUUIDs": [], "txPowerLevel": -2147483648}, "id": "4D:07:36:9B:5A:CE", "name": null, "rssi": -65}