innoveit/react-native-ble-manager #257: Question about pull request

melkayam92, 2017-12-13:
Hey,

I had a problem where sometimes the connection was somehow lost - but on the Android Bluetooth Manager, if I asked for all connected devices - I see it. just to demonstrate:
  1. Connect to a device.
  2. Reload the JS.
  3. The library thinks there are no connections when in fact if you ask for Android Bluetooth Manager for the list of connected devices you will get one. (which results that you can't find the device in scan).

So I fixed it by creating another class that gathers the GATT connections that still connected, and I added another react method to clean all old connected devices - by checking with Android Bluetooth Manager and asking for the class I created for all the gatt connections and just disconnect from them.
 
I don't know if it's a bug, so I'm not sure if I need to pull request this.

Tell me what you think.

marcosinigaglia, 2017-12-15:
Hi, does the `getConnectedPeripherals` method not working?

melkayam92, 2017-12-21:
Hey, 

Getconnectedperipherals return the devices that the library holds in the collection, in the scenrio I described the collections will be empty.

cheehieu, 2018-01-09:
Hi @melkayam92, 

I've also experienced certain situations where the list of connected peripherals in the library differs from the complete list of GATT connections. I'm interested in using your new class functionality to perform a mass disconnect so my app can run from a known state of no connections, then run a scan to find a complete list of available peripherals. Can you share your code? Does it also work for iOS?

Thanks!

melkayam92, 2018-01-09:
Hey @cheehieu,

Yeah sure tomorrow I will post the code here.

Well, I only had problems with Android so the fix only for Android.

You have same issues with iOS?

cheehieu, 2018-01-11:
Thanks @melkayam92!

I just tested with iOS and it seems like it is working fine. Sorry, for some reason I thought it was a bug for both Android/iOS... My iOS app was able to use `BleManager.getConnectedPeripherals()` successfully to return an existing connection, even though that connection was established using a different app (not using the library). However, this didn't work for me on Android.

It would be great to see how you are handling this!

melkayam92, 2018-01-11:
@cheehieu 
My solution will only disconnect the devices you connected from your app.
In Android you can't disconnect from a device that you don't have the GATT connection. 

It will still help you?

marcosinigaglia, 2018-01-11:
Hi, the 6.2.6 release fix the problem in Android.

closed 2018-01-11 by marcosinigaglia

cheehieu, 2018-01-11:
Thanks @marcosinigaglia! I just confirmed that `getConnectedPeripherals()` is now working for me on Android with the v6.2.6 release. 

@melkayam92 Thanks for the insights. I'll need to figure out some other workaround to manage those connections.