innoveit/react-native-ble-manager #258: "BleManagerStopScan" and "BleManagerDiscoverPeripheral"

peiyc, 2017-12-14:
"dependencies": {
		"buffer": "^5.0.8",
		"iconv-lite": "^0.4.19",
		"react": "16.0.0",
		"react-native": "0.50.4",
		"react-native-ble-manager": "^6.2.4",
		"react-native-storage": "^0.2.2",
		"react-redux": "^5.0.6",
		"redux": "^3.7.2",
		"redux-thunk": "^2.2.0",
		"stream": "0.0.2"
	},
	"devDependencies": {
		"babel-jest": "21.2.0",
		"babel-preset-react-native": "4.0.0",
		"jest": "21.2.1",
		"react-test-renderer": "16.0.0",
		"redux-devtools": "^3.4.1",
		"redux-logger": "^3.0.6"
	}

I run it on Android.
I listened "BleManagerStopScan" and "BleManagerDiscoverPeripheral" , but it trigged "BleManagerStopScan" first before "BleManagerDiscoverPeripheral"  when I start scan, and it Re trigged "BleManagerStopScan" when scanning for peripherals is ended.

An other question is I found is after it connected bluetooth device,  I turn off the bluetooth, then I turn on the bluetooth it can not trigger "BleManagerDiscoverPeripheral"  until it trigged "BleManagerStopScan" I can not found buletooth device.

marcosinigaglia, 2017-12-15:
Sorry, I don't understand the problem. Do the BleManagerDiscoverPeripheral triggered always after the BleManagerDiscoverPeripheral?

peiyc, 2017-12-27:
@marcosinigaglia I`m sorry, my English is soso.

Look at this:

I connected the device.It`s no problem.
[image]

After connected  I turn off the bluetooth, and disconnect the device.
[image]

Then I turn on the bluetooth, I can not find the bluetooth.
[image]

marcosinigaglia, 2017-12-28:
If you turn off the bluetooth is strange that you don't find the peripheral, I have to test on Android.

peiyc, 2018-01-06:
After I turn off the bluetooth, I turn on and scan again, but I can`t find the peripheral.

closed 2018-01-11 by marcosinigaglia