innoveit/react-native-ble-manager #259: Characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 not found.

Anandks1993, 2017-12-15:
### Version

Tell us which versions you are using:

- react-native-ble-manager ^6.2.4
- react-native 0.50.4
- Android 7.0

### Expected behaviour

I tried to read the data from sensor through BLE using read method. Peripheral got connected.

### Actual behaviour

But I got an error saying Characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 not found. The characteristicUUID given here is correct, I cross checked with the device.

marcosinigaglia, 2017-12-15:
Hi, did you call the `retrieveServices`method before read?

Anandks1993, 2017-12-18:
Hi @marcosinigaglia sorry, I forgot to call the `retrieveServices`, I will try with that. Thanks.

Anandks1993, 2017-12-18:
Thank you @marcosinigaglia It worked after calling `retrieveServices`. Following is the response I got from `retrieveServices`. Which serviceUUID and characteristicUUID I need to pass to `read` ? Because I see lot of characterisctics and services. 

```{
  "characteristics": [
    {
      "properties": {
        "Indicate": "Indicate"
      },
      "characteristic": "2a05",
      "service": "1801"
    },
    {
      "properties": {
        "Read": "Read"
      },
      "characteristic": "2a00",
      "service": "1800"
    },
    {
      "properties": {
        "Read": "Read"
      },
      "characteristic": "2a01",
      "service": "1800"
    },
    {
      "properties": {
        "Read": "Read"
      },
      "characteristic": "2aa6",
      "service": "1800"
    },
    {
      "descriptors": [
        {
          "value": null,
          "uuid": "2902"
        }
      ],
      "properties": {
        "Notify": "Notify",
        "Indicate": "Indicate",
        "Write": "Write",
        "Read": "Read"
      },
      "characteristic": "beb5483e-36e1-4688-b7f5-ea07361b26a8",
      "service": "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
    }
  ],
  "services": [
    {
      "uuid": "1801"
    },
    {
      "uuid": "1800"
    },
    {
      "uuid": "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
    }
  ],
  "rssi": -70,
  "advertising": {
    "bytes": [
      2,
      1,
      6,
      8,
      9,
      77,
      121,
      69,
      83,
      80,
      51,
      50,
      2,
      10,
      235,
      5,
      18,
      32,
      0,
      64,
      0,
      2,
      1,
      6,
      8,
      9,
      77,
      121,
      69,
      83,
      80,
      51,
      50,
      2,
      10,
      235,
      5,
      18,
      32,
      0,
      64,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
    ],
    "data": "AgEGCAlNeUVTUDMyAgrrBRIgAEAAAgEGCAlNeUVTUDMyAgrrBRIgAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=",
    "CDVType": "ArrayBuffer"
  },
  "id": "30:AE:A4:30:BC:2E",
  "name": "MyESP32"
}

marcosinigaglia, 2017-12-18:
It depends on your device, probably is the latest (beb5483e-36e1-4688-b7f5-ea07361b26a8).

Anandks1993, 2017-12-18:
Okay Thank you, I used what you said and it worked fine so I am closing this issue, but I need to take the characteristic and service dynamically(like `id` and `name` of the device), is there any way ?

closed 2017-12-18 by Anandks1993

marcosinigaglia, 2017-12-18:
No, some services/characteristic are standards and the others are defined by the device developers.

Anandks1993, 2017-12-18:
Okay, Thank you so much @marcosinigaglia 😃