innoveit/react-native-ble-manager #260: `allowDuplicates` has no impact on iOS

rynop, 2017-12-15:
### Version

- react-native-ble-manager v6.2.4
- react-native v0.44.0
- iOS v11.2

### Expected behaivor

Setting `allowDuplicates` to `false` on `BleManager.scan()` should not emit the event `BleManagerDiscoverPeripheral` when a Peripheral has already been discovered.

### Actual behaviour

`BleManager.scan([], 5, false);` causes the `BleManagerDiscoverPeripheral` event to be fired for the same Peripheral many times.

### Steps to reproduce

Handle the `BleManagerDiscoverPeripheral` event, log out the discovered peripheral, you will see duplicates.

marcosinigaglia, 2017-12-15:
Hi, the iOS part for the filter is only a parameter so I don't thinks is a module problem.

rynop, 2017-12-15:
@marcosinigaglia thanks for the quick reply.  I'm not quite sure what you mean - are you saying the `allowDuplicates` flag is just a flag that is passed on directly to the Objective C / Swift code?  And therefore since you are just passing along the value to the native code, its not a problem with `react-native-ble-manager`?

marcosinigaglia, 2017-12-15:
Hi @rynop , yes is only a parameter passed to the Objective C code, try to search for any problem with CBCentralManagerScanOptionAllowDuplicatesKey and your ble device.

closed 2017-12-28 by marcosinigaglia