innoveit/react-native-ble-manager #265: IOS notification error

zjsczzt, 2018-01-08:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.44.3
- iOS/Android v.ios

### Expected behaviour

android and ios can communicate with ble

### Actual behaviour
only android can communicate with ble 

run in android can open this notification, but run in ios can't open this notification.
ffe0 and ffe4 is my chose nofity.in android this can be open ,but in ios it't not working.

### Steps to reproduce

1.
2.
3.

### Stack trace and console log
nofityServiceUUID 0000FFE0-0000-1000-8000-00805F9B34FB nofityCharacteristicUUID 0000FFE4-0000-1000-8000-00805F9B34FB
BleModule.js:248 Notification error: Could not find service with UUID 0000FFE0-0000-1000-8000-00805F9B34FB on peripheral with UUID B71449BD-4F8E-A526-4381-C410A3F9E198

```
```

marcosinigaglia, 2018-01-10:
Try with an app like LightBlue if the notification is working.

zjsczzt, 2018-01-11:
LightBlue can working. but in react-native app  ,the same of notification can't working

marcosinigaglia, 2018-01-11:
Try to add a delay after the `retrieveServices` call.

closed 2018-01-15 by marcosinigaglia

thomastao0215, 2018-02-06:
I have the same issue.Plz give the solution

melkayam92, 2018-02-09:
I also had problems with notification on iOS. 

On different versions for iOS it's worked differently.. but in the end, the hardware needed to implement a descriptor notification in order to make it work. so I'm not sure it's a problem of the library.

Also, try to turn off and on the Bluetooth - that will clean the cache.

nagaokashi, 2018-09-10:
I got the solution.
Use short id 'FFE1' instead of '0000FFE1-0000-1000-8000-00805F9B34FB';

nikoWah, 2018-10-19:
@nagaokashi Thanks, you're a life saver. Was facing the same problem. :)