innoveit/react-native-ble-manager #266: More of a feature then a issue

folofse, 2018-01-11:
### Version

Tell us which versions you are using:

- react-native-ble-manager v. 6.2.6
- react-native v. 0.52
- iOS/Android v. iOS 11

### Expected behaviour

I creating a Bedmotor app for one of my clients. The issue is that I need repeat write commands within 500ms in order to get the motor to run smooth. I tried to write with response, and when the callback returns I write the next command, but I takes to long time for the data to return and the motor begins to stutter. 

I also tried to write without response, but I got a warning that iOS will use with response anyway. 

Last I tried to write with a timeout on 500ms, but then the que lined up and the motor was moving long after I released the buttons. 

Maybe its possible to add a flag to write method and tell it to bail early and not wait for "didWriteValueForCharacteristic" to occur?

Any thoughts on this?

Regards Olof

marcosinigaglia, 2018-01-11:
Hi @folofse, is strange that the write without response is not working. What is the warning?

folofse, 2018-01-11:
Hi!

[CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x1c02b99e0, UUID = 01AC, properties = 0xA, value = (null), notifying = NO> does not specify the "Write Without Response" property - ignoring response-less write

So it won't write

marcosinigaglia, 2018-01-11:
Is the ble device that not support the "Write Without Response"

folofse, 2018-01-11:
Hmm, maybe not. I'm not 100% sure.  According to this, the pheripherals will ignore "Write Without Response" 

https://stackoverflow.com/questions/25672376/ios-ble-write-without-response-property-ignoring-response-less-write 

The problem is that the response time is slower then the motors update rate.

marcosinigaglia, 2018-01-11:
Yes, to speed up you need to use writeWithoutResponse, is better if you ask to change the ble device configuration to support it.

folofse, 2018-01-11:
Yes I will do that, thank you for your support!

closed 2018-01-12 by marcosinigaglia