innoveit/react-native-ble-manager #269: Android7.1No scan of Bluetooth

moma555, 2018-01-16:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2018-01-22:
Hi, is impossible to understand somethings with this kind of issue.

closed 2018-01-22 by marcosinigaglia

moma555, 2018-01-22:
I'm really sorry, I found that the device was a X86 structure, so I couldn't scan it and thank you.