innoveit/react-native-ble-manager #270: Android is hanging on connect

rchaider, 2018-01-19:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v.0.48.0
-Android 8.1

### Expected behaviour
On iOS, I can connect just fine to using both my app and your sample app to my bluetooth device. It's a BLE18&22 dual-mode bluetooth  module. Works on BLE and SPP.

When I run your sample app and do a scan, I see it listed. Tapping on it however causes it to just  hang on .connect(). The connect never resolves. It sits and sits until eventually I get:

> Disconnected from 1E:04:E0:81:4D:7B
> Connection error Connection error

### Tell us what should happen
It should connect and move on to retrieve services but it's not able to.

I have tried restarting my phone, toggling bluetooth on / off, etc. I can use a bluetooth explorer app on my phone to connect to the device and send / receive commands. It's just not working with the blemanager.

```
```

marcosinigaglia, 2018-01-22:
Hi, sorry I never tested on Android 8, you have to try to debug the connection method.

garbo-a, 2018-02-01:
The same issue is here https://github.com/innoveit/react-native-ble-manager/issues/252

closed 2018-06-14 by marcosinigaglia

bijoycse, 2019-06-07:
How APP is sending command(0x11) to my ble device by using react-native-ble-manager???