innoveit/react-native-ble-manager #272: 为什么不添加reactMixin(App.prototype, TimerMixin);，就不能接收数据？

longmingfeng, 2018-01-22:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2018-01-22:
English please.

stanleywijoyo, 2018-01-22:
Not sure if this is the same issue, but after compiling the example code on my android phone, I could only see the peripheral IDs. All the peripheral names are null.

**Version**

- react-native-ble-manager v6.2.6
- react-native v0.50.3
- Android 7.1.1

marcosinigaglia, 2018-01-23:
@longmingfeng 
1) Yes but the ids are different in iOS and Android
2) I think you made other error in the changes.

closed 2018-01-24 by marcosinigaglia