innoveit/react-native-ble-manager #273: iOS Application in Test Flight/release build doesn't detect expected peripheral devices

t-roberts, 2018-01-22:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.4
- react-native v0.48.4
- iOS 9.1

### Expected behaviour

When connected to a PC and running the app through xcode, my application can start, scan, detect, and connect to the expected peripheral devices

### Actual behaviour

When i build the application and publish the app to test flight, the scan does not detect the expected peripheral devices.  Am I missing something in the settings for apps in release mode instead of debug mode?

t-roberts, 2018-01-22:
Update:  When running the application in release mode from xcode, this error occurs, and does not occur in debug:
CoreBluetooth[API MISUSE] can only accept commands while in the powered on state

Any idea what may be causing this?

marcosinigaglia, 2018-01-23:
I think the bluetooth is off, you have to check before.

t-roberts, 2018-01-23:
Bluetooth is definitely on.  Like i said, this exact code works when tethered to a machine in debug, but fails in release.  I've also added the checkState call to make sure that the phone is reading that ble is ON.

whitedogg13, 2018-01-23:
@t-roberts Interesting issue, maybe related to JS runtime. Can you further check whether app works correctly for both debugger on/off cases? 

One fact often be ignored by RN developers is: 
> when in release (or non-debugging) mode, the JavaScriptCore in iOS will be the JS runtime, on the other hand for debugging, the V8 in Chrome will be the JS runtime.  

BTW, this kind of issue is totally possible. Per my experience, `iOS 9.x` does in some case use `array-like` objects (ex. `UINT8Array`) rather than normal `array`, which will result in `TypeError` when you want to invoke array method such as `forEach` or `map`.

Hope that helps!

t-roberts, 2018-01-23:
@whitedogg13 thanks for the reply.  This code does work with both debugger on/off, in debug mode.

Shauneoo, 2018-01-24:
I'm having the same issue:

`2018-01-24 19:26:10.808302+0000 ****[10860:1204154] BleManager created
2018-01-24 19:26:11.022258+0000 ****[10860:1204211] BleManager initialized
2018-01-24 19:26:11.024337+0000 ****[10860:1204211] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0073f40> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
2018-01-24 19:26:11.045930+0000 ****[10860:1204219] scan with timeout 3
2018-01-24 19:26:11.046050+0000 ****[10860:1204219] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0073f40> can only accept this command while in the powered on state
2018-01-24 19:26:14.064273+0000 ****[10860:1204154] Stop scan
2018-01-24 19:26:14.064754+0000 ****[10860:1204154] [CoreBluetooth] API MISUSE: <CBCentralManager: 0x1c0073f40> can only accept this command while in the powered on state`

RN version: 0.52.1,
react-native-ble-manager: 6.2.6

t-roberts, 2018-01-25:
Any updates or thoughts on this?

marcosinigaglia, 2018-01-25:
Sorry, I'm not able to simulate your problm, try to use the `restoreIdentifierKey` option in the start method.

t-roberts, 2018-01-25:
So, including this in the startup did get the iOS device to scan and recognize devices.  However, The .connect method does not return a promise or an error.  I can see that i am connecting to the device, but theres no acknowledgement coming back from the connect method.

Shauneoo, 2018-01-25:
@t-roberts I'm still testing, but enabling Target > Background Modes > Uses Bluetooth LE accessories appears to have resolve my issue.

t-roberts, 2018-01-31:
got this to work by enabling LE Accessories and also providing a restoreIdentifierKey.

closed 2018-01-31 by t-roberts

adnanba, 2018-05-30:
@t-roberts 
Can you please provide more info. I have enabled LE Accessories and provided a restoreIdentifierKey, but I'm still getting the same error message API MISUSE

Alexxzz, 2018-07-09:
For me the fix for the `[CoreBluetooth] API MISUSE: <CBCentralManager: 0x...> can only accept this command while in the powered on state` was to wait for a device to actually start Bluetooth by adding a delay after `BleManager.start()` and before calling `BleManager.scan(...)`.

Probably using the `BleManagerDidUpdateState` event would be a better solution, but did not tried it yet.

sishpalR, 2024-08-01:
Below code working with me.

let data = dataByteArray.withUnsafeBufferPointer { bufferPointer -> Data in
return Data(buffer: bufferPointer)
}
myPeripheral?.writeValue(data, for: firstWriteCharacteristic, type: .withResponse)

Remove the warning from the data object when creating data to be passed to the writeValue method.