innoveit/react-native-ble-manager #275: Get bonded devices and start notification on iOS issues

ducNgbk, 2018-01-25:
Hello, I'm developing app which uses secure connection
There are some missing features that I'm wondering if it's possible to add on iOS
- get boned device list
- start notification return specific error if bond is needed

Thank you, guys, for your awesome work

marcosinigaglia, 2018-01-25:
Hi, I don't have a device to use to develope this things so a PR is welcome.

whitedogg13, 2018-04-11:
Hi, as far as I know iOS doesn't expose any public API to get bonded devices, you can see discussions like [this SO link](https://stackoverflow.com/questions/48361435/ios-previously-bonded-ble-devices?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa). So currently it's not possible to support this feature.

PS: I have committed the Android `getBondedPeripherals` for this library, and also searching solutions for iOS but sadly cannot find one~

closed 2018-04-11 by marcosinigaglia