innoveit/react-native-ble-manager #276: Can't disconnect from device, isPeripheralConnected returning correct value

nivnahmias, 2018-01-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.6
- react-native v0.49.3
- Android v.?

After scanning for devices, i'm able to connect to my QC35 headphones.
Calling isPeripheralConnected('xxx',[]) returns false.
Calling getConnectedPeripherals([]) returns the headphones and i can't seem to remove them.

Am i using this library incorrectly?

marcosinigaglia, 2018-01-25:
Hi, I think isPeripheralConnected need some fix. I don't understand the second problem, are you not able to disconnect?

nivnahmias, 2018-01-25:
Yeah, calling "removePeripheral" returns a rejected promise and the device just stays connected.

marcosinigaglia, 2018-01-25:
You have to use the `disconnect` method.

nivnahmias, 2018-01-25:
i'm sorry i meant calling `disconnect`.
after calling `disconnect`, calling `getConnectedPeripherals` still returns my headphones.
only after manually turning off my headphones or disconnecting them through the bluetooth settings i get the correct answer from `getConnectedPeripherals`.

vpfaulkner, 2018-05-16:
I think I'm running into this issue as well @marcosinigaglia. After shutting down the process acting as the peripheral, if I call `isPeripheralConnected` then `true` is still returned. Also the `BleManagerDisconnectPeripheral` did not get triggered. Am I missing something?

diablourbano, 2018-07-10:
Same error, I'm even calling `removePeripheral` (I'm using an android device) and if I run `getConnectedPeripherals` I'll get the device, and I'll get `true` if I run `isPeripheralConnected`.

NN01010101, 2018-10-26:
Can confirm exact same issue!

yoshimoto8, 2018-11-08:
Same error.  it is not correct value when i call `isPeripheralConnected` before `disconnected`

PR #410 (2018-11-27): Fill the internal peripherals map in the getConnectedPeripherals method
The goal was to be able to use or disconnect from an already connected BLE device after an app restart/reload (without scanning) on Android. Previously these always resulted in a "Peripheral not found" error.

- Changed the map's visibility to private and moved Peripheral creation logic inside BleManager to make it cleaner. And used this savePeripheral method inside getConnectedPeripherals. 
- Added synchronized blocks for every modification of the map. These might not be necessary but a LinkedHashMap does not guarantee th [... cut]
closed 2018-11-27 by marcosinigaglia