innoveit/react-native-ble-manager #281: Once connected, what then?

NaseebullahSafi, 2018-02-02:
The thing is, I understand the concept of BLE GATT and how it works, but I don't actually know how to implement what I require.

For example, my Arduino is constantly sending out data, so when my Samsung device is connected to Arduino, I would like to receive that data.

But I'm not entirely sure how to initiate or start retrieving the data.

```js
   // starts the scan
   this.startScan();
    
    // successfully connects to my arduino
    BleManager.connect(`${RFDuinoID}`).then(() => console.log("connected") ).catch( e => console.log( e ));
    setTimeout(() => {
      BleManager.retrieveServices(RFDuinoID).then(peripheral => {
        console.log(peripheral);
        let serviceUUID              = "2220";
        let characteristicUUID       = "2221";
        setTimeout(() => {
          BleManager.startNotification(RFDuinoID, serviceUUID, characteristicUUID).then(() => {
            console.log('Started notification on ' + RFDuinoID);
            setTimeout(() => {
              // I'm not entirely sure what I'm doign here
              const data = stringToBytes('b');
              BleManager.read(RFDuinoID, serviceUUID, characteristicUUID, data).then(() => {
              });
            }, 500);
          }).catch((error) => console.log('Notification error', error));
        }, 200)
      })
    }, 900) // <== 1
```

So the main idea is, what do I need to call in order to start getting those data?

marcosinigaglia, 2018-02-05:
Hi, you need to subscribe to `BleManagerDidUpdateValueForCharacteristic` event instead using the read method.

NaseebullahSafi, 2018-02-05:
Hi there, Thank you @marcosinigaglia I'll give it a try, and will let you know if I get anything back. Thanks

NaseebullahSafi, 2018-02-05:
@marcosinigaglia 

I see that we already have an eventlistener for that? Sorry what do you mean by subscribe?

```js
  componentDidMount() {
    AppState.addEventListener('change', this.handleAppStateChange);

    BleManager.start({showAlert: false}).then(() => console.log("module initialised"));

    this.handlerDiscover = bleManagerEmitter.addListener (
      'BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral
    );

    this.handlerStop = bleManagerEmitter.addListener (
      'BleManagerStopScan', this.handleStopScan
    );

    this.handlerDisconnect = bleManagerEmitter.addListener (
      'BleManagerDisconnectPeripheral', this.handleDisconnectedPeripheral
    );

    this.handlerUpdate = bleManagerEmitter.addListener (
      'BleManagerDidUpdateValueForCharacteristic', this.handleUpdateValueForCharacteristic
    );

    this.platformPermissionRequest();

    this.startScan();

    BleManager.connect(`${RFDuinoID}`).then(() => console.log("connected") ).catch( e => console.log( e ));

    setTimeout(() => {
      BleManager.retrieveServices(RFDuinoID).then(peripheral => {
        console.log(peripheral);
        let { serviceUUID , characteristicUUID } = '00002220-0000-1000-8000-00805F9B34FB';

        setTimeout(() => {
          BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
            console.log('Started notification on ' + RFDuinoID);
            setTimeout(() => {
              const data = stringToBytes('b');

            }, 500);
          }).catch((error) => console.log('Notification error', error));
        }, 200)
      })
    }, 900) // <== 1
}
```

NaseebullahSafi, 2018-02-07:
@marcosinigaglia 

I've subscribed as below, but nothing is returned for the subscription. And no error is thrown. Have I missed something?

```js
    this.startScan();

    BleManager.connect(`${RFDuinoID}`).then(() => console.log("connected") ).catch( e => console.log( e ));

    setTimeout(() => {
      BleManager.retrieveServices(RFDuinoID).then(peripheral => {
        console.log("peripheral ", peripheral);
        setTimeout(() => {
          BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
            console.log('Started notification on ' + RFDuinoID);
            setTimeout(() => {
              NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', (data) => {
                console.log("BMDUVFC: ", data.value);
              });
            }, 500);
          }).catch((error) => console.log('Notification error', error));
        }, 200)
      })
    }, 900) // <== 1
```

marcosinigaglia, 2018-02-09:
Hi, are you sure that the device is sending data? Maybe you can use an app like "Ble scanner" to test if the device is working.

NaseebullahSafi, 2018-02-12:
Hi there, sorry to get back so late. 

Basically, you're right, the device isn't sending data because it requires a command to start sending the packets. The following code (using pure Noble) is used in desktop to start sending data:

```js
      this._rfduinoService.once(k.OBCINobleEmitterServiceCharacteristicsDiscover, (characteristics) => {
        if (this.options.verbose) console.log('Discovered ' + characteristics.length + ' service characteristics');
        for (let i = 0; i < characteristics.length; i++) {
          // console.log(characteristics[i].uuid);
          if (characteristics[i].uuid === k.RFduinoUuidReceive || characteristics[i].uuid === rfduinoUuidReceiveLong) {
            if (this.options.verbose) console.log("Found receiveCharacteristicUUID");
            this._receiveCharacteristic = characteristics[i];
          }
          if (characteristics[i].uuid === k.RFduinoUuidSend || characteristics[i].uuid === rfduinoUuidSendLong) {
            if (this.options.verbose) console.log("Found sendCharacteristicUUID");
            this._sendCharacteristic = characteristics[i];
          }
        }

        if (this._receiveCharacteristic && this._sendCharacteristic) {
          this._receiveCharacteristic.on(k.OBCINobleEmitterServiceRead, (data) => {
            // TODO: handle all the data, both streaming and not
            this._processBytes(data);
          });

          // if (this.options.verbose) console.log('Subscribing for data notifications');
          this._receiveCharacteristic.notify(true);

          this._connected = true;
          this.emit(k.OBCIEmitterReady);
          resolve();
        } else {
          reject('unable to set both receive and send characteristics!');
        }
```

But I'm not quite sure how we can tell our device to starts sending data form your library.

marcosinigaglia, 2018-02-12:
Hi, I use Noble only to testing so I'm not expert. The code find two characteristics, one for the notification and the other to write (this._sendCharacteristic), search in the code where is used.

NaseebullahSafi, 2018-02-12:
@marcosinigaglia What's that usually doing is simply check if it exists. I think the issue lies here:

```js
         // k.OBCINobleEmitterServiceRead is "read"
          this._receiveCharacteristic.on(k.OBCINobleEmitterServiceRead, (data) => {
            // TODO: handle all the data, both streaming and not
            this._processBytes(data);
          });
```

Here's the [`full code`](https://github.com/PushTheWorld/OpenBCI_NodeJS_Cyton_BLE/blob/master/openBCICytonBLE.js) at line 763 to see the whole function.

NaseebullahSafi, 2018-02-12:
For some reason, they're able to get some data on that "read" event listener.

marcosinigaglia, 2018-02-12:
I think the line 651 is where they send the "start" command, the methos is `streamStart` so I think is the correct one. Search what is the k.OBCIStreamStart constant e try to write that after the notification.

marcosinigaglia, 2018-02-12:
I think is [that](https://github.com/OpenBCI/OpenBCI_JavaScript_Utilities/blob/421dc163e3122d4a5460b7cee28bc8998cb93875/src/constants.js#L143)

NaseebullahSafi, 2018-02-12:
Hmmm I see. "k.OBCIStreamStart" is 'b'. I'll try doing that and hopefully see what we get.

NaseebullahSafi, 2018-02-12:
Thanks for the heads up. Appreciate it

NaseebullahSafi, 2018-02-13:
@marcosinigaglia I used the ble scanner, and at first nothing was being returned. This is because I first had to write the char 'b' in the bleScanner, and then it started sending the data. In our application, we have the write method, but that method takes in data as byte array. And therefore the write method throws an error saying write failed. the byte array for 'b' is [98].

What is the issue?

NaseebullahSafi, 2018-02-13:
@marcosinigaglia 

This is the code:

```js
    const BCommand = stringToBytes('b');
    BleManager.connect(`${RFDuinoID}`).catch( e => console.log( "Error, could not connect(): ", e ));

    setTimeout(() => {
      BleManager.retrieveServices(RFDuinoID).then(peripheral => {
        console.log("peripheral ", peripheral);
        setTimeout(() => {
          BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
            console.log('Started notification on ' + RFDuinoID);
            setTimeout(() => {
              BleManager.write(RFDuinoID, "2220", "2221", BCommand).then(() => {
                console.log('Write: ', BCommand);
              }).catch(e => console.log(e, {b: BCommand}));
            }, 500);
          }).catch((error) => console.log('Notification error', error));
        }, 200);
      });
    }, 900);
```

marcosinigaglia, 2018-02-13:
From your previous screenshot I see that the characteristic `2221` is not writable.

NaseebullahSafi, 2018-02-13:
Yes that is correct, I had changed that It is now able to write. But every time it writes, I get an error "cannot read property 'apply' of undefined".

```js
    const BCommand = stringToBytes('b');
    BleManager.connect(`${RFDuinoID}`).then(() => {
      setTimeout(() => {
        BleManager.retrieveServices(RFDuinoID).then(peripheral => {
          console.log("peripheral ", peripheral);
          setTimeout(() => {
            BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
              console.log('Started notification on ' + RFDuinoID);
              setTimeout(() => {
                NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', (d) => console.log("data: ", d.value))
                BleManager.write(RFDuinoID, "2220", "2222", BCommand).then(() => console.log('write(): ', BCommand)).catch(e => console.log("Error on write(): ", e));
              }, 500)
            }).catch((error) => console.log('Notification error', error));
          }, 200);

        });
      }, 900);
    }).catch( e => console.log( "Error, could not connect(): ", e ));
```

NaseebullahSafi, 2018-02-13:
Do you think it might refer to not having a read method? @marcosinigaglia

NaseebullahSafi, 2018-02-13:
@marcosinigaglia  Ok So far I've broken down the problem into this:

If the write becomes successful, then the read method is throwing the error for
> Cannot read property 'apply' of undefined

```js
    const BCommand = stringToBytes('b');
    BleManager.connect(`${RFDuinoID}`).then(() => {
      setTimeout(() => {
        BleManager.retrieveServices(RFDuinoID).then(peripheral => {
          console.log("peripheral ", peripheral);
          setTimeout(() => {
            BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
              console.log('Started notification on ' + RFDuinoID);
              setTimeout(() => {
                BleManager.write(RFDuinoID, "2220", "2222", BCommand).then(() => console.log('write(): ', BCommand)).catch(e => console.log("Error on write(): ", e));
              }, 500);
              setTimeout(() => {
                BleManager.read(RFDuinoID, "2220", "2221").then((d) => console.log('read(): ', d)).catch(e => console.log("Error on read(): ", e));
              }, 2000);
            }).catch((error) => console.log('Notification error', error));
          }, 200);

        });
      }, 900);
    }).catch( e => console.log( "Error, could not connect(): ", e ));
```

marcosinigaglia, 2018-02-13:
You have to handle the BleManagerDidUpdateValueForCharacteristic and not use the read.

NaseebullahSafi, 2018-02-13:
Should the `BleManagerDidUpdateValueForCharacteristic ` be handled before the write or after?

NaseebullahSafi, 2018-02-13:
@marcosinigaglia  Ok So I've removed the read method and subscribed to `BleManagerDidUpdateValueForCharacteristic `

```js
    const BCommand = stringToBytes('b');
    BleManager.connect(`${RFDuinoID}`).then(() => {
      setTimeout(() => {
        BleManager.retrieveServices(RFDuinoID).then(peripheral => {
          console.log("peripheral ", peripheral);
          setTimeout(() => {
            BleManager.startNotification(RFDuinoID, "2220", "2221").then(() => {
              console.log('Started notification on ' + RFDuinoID);
              NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', (d) => console.log("data: ", d.value));
              setTimeout(() => {
                BleManager.write(RFDuinoID, "2220", "2222", BCommand).then(() => console.log('write(): ', BCommand)).catch(e => console.log("Error on write(): ", e));
              }, 2000);
            }).catch((error) => console.log('Notification error', error));
          }, 200);
        });
      }, 900);
    }).catch( e => console.log( "Error, could not connect(): ", e ));
```

But I still get the same error:
[image]

marcosinigaglia, 2018-02-13:
I think you have some error in the code.

NaseebullahSafi, 2018-02-13:
I have absolutely no idea where I'm going wrong. As far as I know the write method is being called at the right place. @marcosinigaglia

marcosinigaglia, 2018-02-13:
Have you checked the ....ptionsManager.js file?

NaseebullahSafi, 2018-02-13:
Well, that file is something not in my project folder. And it I can't seem to debug using that file.

closed 2018-03-08 by marcosinigaglia