innoveit/react-native-ble-manager #287: I can find only MacBook

vladinko0, 2018-02-13:
### Version

Tell us which versions you are using:

- react-native-ble-manager v 6.2.6
- react-native v 0.53
- Android v. 7.0

### Expected behaviour

I would like to find by scanningg all available devices: iPhone, Desktop, other Android, Medical device, MacBook.

### Actual behaviour

I can find only MacBook.

### Steps to reproduce

1. Using Example from react-native-ble-manager/example/App.js
2. Turning on Bluetooth on all devices.
3. Tapping 'Scan Bluetooth' button in application.

marcosinigaglia, 2018-02-13:
You see only the visible bluetooth LE devices.

vladinko0, 2018-02-13:
Is it a big problem to implement classic bluetooth devices?

marcosinigaglia, 2018-02-23:
It's a different thing, search for others libraries.

closed 2018-02-23 by marcosinigaglia