innoveit/react-native-ble-manager #288: Crash on DidUpdateValueForCharacteristic

eeynard, 2018-02-14:
Hi,

### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.6
- react-native v.0.52
- iOS/Android iOS 10.3

### Expected behaviour

I am implementing a communication protocol that write several bytes to a BLE device and expect a pretty long answer. 

### Actual behaviour

I'm starting a notification and waiting for the updated values. It works fine for a while but in the middle of the exchange (several DidUpdateValueForCharacteristic in a row), the app crash on iOS (works fine on android).

Tell us what happens instead

### Steps to reproduce

1. Connect to a BLE device
2. Start a notification
3. Expect to receive a lot of updated values

### Stack trace and console log

Here is the console trace
```
2018-02-14 17:23:34:664 ble-app[1241:1131570] Expo Remote Notification services won't work in an ExpoKit app because Expo can not manage your APNS certificates.
2018-02-14 17:23:34:807 ble-app[1241:1131570] 2018-02-14 17:23:34.803 [warn][tid:main][RCTBridge.m:120] Class RCTDevMenu was not exported. Did you forget to use RCT_EXPORT_MODULE()?
2018-02-14 17:23:34:816 ble-app[1241:1131570] 2018-02-14 17:23:34.815 [warn][tid:main][RCTBridge.m:120] Class RCTCxxModule was not exported. Did you forget to use RCT_EXPORT_MODULE()?
2018-02-14 17:23:34:820 ble-app[1241:1131570] 2018-02-14 17:23:34.821 [warn][tid:main][RCTBridge.m:120] Class EXTest was not exported. Did you forget to use RCT_EXPORT_MODULE()?
2018-02-14 17:23:34:822 ble-app[1241:1131570] 2018-02-14 17:23:34.823 [warn][tid:main][RCTBridge.m:120] Class EXFrameExceptionsManager was not exported. Did you forget to use RCT_EXPORT_MODULE()?
2018-02-14 17:23:34:925 ble-app[1241:1131570] 2018-02-14 17:23:34.925 [warn][tid:main][RCTModuleData.mm:69] Module EXGLObjectManager requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.
2018-02-14 17:23:34:927 ble-app[1241:1131570] 2018-02-14 17:23:34.927 [warn][tid:main][RCTModuleData.mm:69] Module EXGLViewManager requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.
2018-02-14 17:23:34:935 ble-app[1241:1131570] 2018-02-14 17:23:34.935 [warn][tid:main][RCTModuleData.mm:69] Module RCTImageLoader requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.
2018-02-14 17:23:38:762 ble-app[1241:1131570] 2018-02-14 17:23:38.762 [info][tid:main][RCTCxxBridge.mm:210] Initializing <RCTCxxBridge: 0x156c02e0> (parent: <RCTBridge: 0x156fb210>, executor: (null))
2018-02-14 17:23:38:769 ble-app[1241:1131570] 2018-02-14 17:23:38.770 [warn][tid:main][RCTModuleData.mm:69] Module EXGLObjectManager requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.
2018-02-14 17:23:38:771 ble-app[1241:1131570] 2018-02-14 17:23:38.771 [warn][tid:main][RCTModuleData.mm:69] Module EXGLViewManager requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.
2018-02-14 17:23:38:777 ble-app[1241:1131570] 2018-02-14 17:23:38.777 [warn][tid:main][RCTModuleData.mm:69] Module RCTImageLoader requires main queue setup since it overrides `init` but doesn't implement `requiresMainQueueSetup`. In a future release React Native will default to initializing all native modules on a background thread unless explicitly opted-out of.

[...]

libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) ¨
```

And here is the piece of code where the crash happened
[image]

Do you have any clue on how to fix it or at least how I could have more useful info about the crash ?

Thanks !

marcosinigaglia, 2018-02-15:
Are you using Expo? I don't think you can use external module.

eeynard, 2018-02-15:
Yes but I detached (to be able to use external modules). And it's actually working pretty well on both platforms except some specific crash like this one.

marcosinigaglia, 2018-02-15:
Sorry, you have to debug in Xcode

eeynard, 2018-02-15:
That is what I am trying to do but I am kind of a rookie on xcode. After struggling for a while I finally achieve to activate system logs (there was a bug with iOS 10 physical devices) and I ran into a more detailed error (I will update my first message) :
```
 *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[3]'
```

eeynard, 2018-02-15:
I found a fix to this issue. I will push a PR and link it here.

PR #289 (2018-02-15): Fix null updateValue crash on iOS #288
Handle empty values.
Fix #288
closed 2018-02-15 by marcosinigaglia