innoveit/react-native-ble-manager #291: startNotification doesn't enable NOTIFY property

simone201, 2018-02-20:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.6
- react-native v0.53.0
- Android 7.0

### Expected behaviour

When calling startNotification, as i can see from the native code (Android in particular, but i hope also iOS), the library should enable the NOTIFY char configuration descriptor value to 1 so the BLE device will start notifying data.

### Actual behaviour

This doesn't happen, and the BLE device stays with NOTIFY set to 0. If we enable first notify via another app then it works as it is already enabled.

### Steps to reproduce

1. Connect to BLE device, retrieve services, make sure Notify is disabled by default in ble device
2. Call startNotification on any service/char which has notify disabled by default
3. No notifications incoming

### Stack trace and console log

No error is thrown, no stacktrace either

```
```

Hope to get some help, thanks in advance!

marcosinigaglia, 2018-02-20:
Hi @simone201 , I think you have to debug the `setNotify` method https://github.com/innoveit/react-native-ble-manager/blob/ee12ea068f31fb95adfbb2449b683ec5ba8e69dc/android/src/main/java/it/innove/Peripheral.java#L385

simone201, 2018-02-22:
`
02-22 11:20:37.079 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.079 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.079 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.080 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a00b-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.082 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a00b-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.085 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify complete
02-22 11:20:37.086 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.086 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.086 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.087 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a000-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.090 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a000-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.091 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.091 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.091 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.091 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a002-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.094 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a002-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.095 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.095 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.095 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.095 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a003-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.098 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a003-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.098 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.099 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.099 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.099 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a004-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.103 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a004-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.104 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.104 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.104 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.105 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a005-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.108 1518-5911/it.varsirioindustria.tpmsmanager D/logs: startNotification
02-22 11:20:37.108 1518-5911/it.varsirioindustria.tpmsmanager D/logs: registerNotify
02-22 11:20:37.108 1518-5911/it.varsirioindustria.tpmsmanager D/logs: setNotify
02-22 11:20:37.108 1518-5911/it.varsirioindustria.tpmsmanager D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000a00a-0000-1000-8000-00805f9b34fb enable: true
02-22 11:20:37.111 1518-5911/it.varsirioindustria.tpmsmanager D/logs: Characteristic 0000a00a-0000-1000-8000-00805f9b34fb set NOTIFY
02-22 11:20:37.146 1518-1531/it.varsirioindustria.tpmsmanager D/logs: onDescriptorWrite 0000a00b-0000-1000-8000-00805f9b34fb
`

Just debugged the code as you mentioned, and added a line for onDescriptorWrite as the problem is there, as you can see from the logcat i'm starting the notification for several chars but only the "a00b" gets the descriptor correctly written while the others sometimes don't even get the right descriptor.

What could it be? I'm seriously struggling at this, as all the other apps work like a charm (aka nRF Connect) at enabling chars notifications.

marcosinigaglia, 2018-02-23:
Hi, are you trying to enable the notification in some characteristics?

simone201, 2018-02-23:
Yes, of course, here's my snippet that does the job and enables notification over some characteristics in a cascade fashion:

```javascript
Object.values(charsUUIDs).forEach(async (uuid) => {
    await BleManager.startNotification(device.id, serviceUUID, uuid);
});`
```

and as the logcat says and RN debug says, there is no "visible" failure, but still the chars notifications aren't really enabled.

marcosinigaglia, 2018-02-23:
Ok, the problem is that the module don't have a queue for the operation and you can't send multiple commands simultaneously. You have to start the notification and wait the callback until start with the second operation.

marcosinigaglia, 2018-02-23:
I prefer to handle the queue in the app, but if you want to try any PR is welcome.

closed 2018-02-23 by marcosinigaglia

ydykevin, 2018-09-04:
@simone201 

Hi, I am having the same problem that startNotification might get stuck on Android, maybe it is because I proceed to next operation before the callback.

Can you show me how you do multiple operations using then. ?

drash-course, 2018-11-08:
If someone if still interested in this issue, here is a sample to enable multiple notifications on Android.

`
async startNotifications(deviceID) {
  try {
    await BleManager.startNotification(deviceID, ServiceDemo, Characteristic1);
    log('Did register notifications for char 1');
    await BleManager.startNotification(device.id, ServiceDemo, Characteristic2);
    log('Did register notifications for char 2');
    await BleManager.startNotification(device.id, ServiceDemo, Characteristic3);
    log('Did register notifications for char 3');
  } catch (error) {
    warn('Could not start notifications for all characteristics:', error);
  }
}
`

Girum, 2020-11-29:
> If someone if still interested in this issue, here is a sample to enable multiple notifications on Android.
> 
> ```js
> async startNotifications(deviceID) {
>   try {
>     await BleManager.startNotification(deviceID, ServiceDemo, Characteristic1);
>     await BleManager.startNotification(device.id, ServiceDemo, Characteristic2);
>     await BleManager.startNotification(device.id, ServiceDemo, Characteristic3);
>   } catch (error) {
>     warn('Could not start notifications for all characteristics:', error);
>     // don't forget to throw the error again to reject the promise
>     throw error;
>   }
> }
> ```
> 
> If you want to use the `then` style, you can do this instead.
> 
> ```js
> startNotifications(deviceID) {
>   return BleManager.startNotification(deviceID, ServiceDemo, Characteristic1)
>     .then(() => BleManager.startNotification(deviceID, ServiceDemo, Characteristic2))
>     .then(() => BleManager.startNotification(deviceID, ServiceDemo, Characteristic3))
>     .catch(error => {
>       warn('Could not start notifications for all characteristics:', error);
>       // don't forget to throw the error again to reject the promise
>       throw error;
>     });
> }
> ```

Hi! How can you render the notification data in the screen without load and "freeze"the screen because the re-render?