innoveit/react-native-ble-manager #293: can not write bytes array where bytes array s length is over 20

iOS-mamu, 2018-02-27:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.6
- react-native v2.0.1
- iOS/Android  android 6.0

### Expected behavior

I am writing bytes array to Bluetooth, bytes array s length is 21.
I should write 21 length bytes array successfully.

### Actual behavior

previous 20 is good, but I cannot write the 21st (while reading is good).However, I use my iOS native Bluetooth code can write and read 21th.

Tell us what happens instead

### Steps to reproduce

1.BleManager.write, bytes array s length is 21.
2.when I read the 21st bytes never change.

marcosinigaglia, 2018-02-27:
Hi, the max size for a write is 20 bytes so in your case the library split in two writes (20 bytes and 1 byte).

iOS-mamu, 2018-02-27:
I get it.thank you so much

closed 2018-02-27 by marcosinigaglia