innoveit/react-native-ble-manager #295: How to stop reading data from device ?

Anandks1993, 2018-03-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.? 6.2.6
- react-native v?.? 0.53.3
- iOS/Android v.? Android

### Expected behaviour

I want to stop reading data from device
### Actual behaviour

I tried setting interval for reading data and clearing it but since I am using event emitters to send data from one component(page) to another, I couldn't able to stop it. But if I use the data in same component the interval thing is working fine.

Is there any method or way to stop reading data from device ?

marcosinigaglia, 2018-03-02:
Hi, sorry but I don't understand you problem.

Anandks1993, 2018-03-02:
Forget all the event emitters, I removed that now. I just want to stop reading data from device. Currently I am using setInterval method to read data continuously and stop reading it using clearInterval. Is that fine ? Because sometimes its lagging in performance.

marcosinigaglia, 2018-03-02:
If you can't use the notification is correct. The performance are very different in debug.

Anandks1993, 2018-03-02:
So using notification we can stop reading data ?? Correct me if am wrong

marcosinigaglia, 2018-03-02:
If you use notification is the device that send you the data when is changed so you don't have to request it every time.

Anandks1993, 2018-03-02:
Thank you, I will try that way ☺️

closed 2018-03-06 by Anandks1993