innoveit/react-native-ble-manager #300: Some mobiles wont find the Ble device

gkothe, 2018-03-09:
- react-native-ble-manager v6.2.8
- react-native v0.54.0
- iOS/Android v.7.1.1

I made a blank app with the latest versions of everything , it opens a blank screen and start scanning, when it finds, its prints on console.
So, there are two mobiles that I tested that did not find the expected device. But did find other(s) device
lg-v500
Moto G5S plus
[image]

I also downloaded the "ble checker" from google play to see if it was a hardware problem, but its not (pic1) . 
That app found the device in question (kafer)

[image]

Is that peripheral mode relevant to something? (pic1)

In others mobiles it worked well (tested around 5 different android cellphones ).

Any Ideias?

Thanks.

gkothe, 2018-03-12:
Update: If i have both apps running at the same time, The device is found. (ble scanner and my app)

marcosinigaglia, 2018-03-12:
Hi, do you use the forceLegacy flag?

gkothe, 2018-03-12:
Not using (false), since I was not sure what was that about. Going to try with "true".

nomoreboredom, 2018-05-09:
I was having the same issue as @gkothe, and using forceLegacy flag solved the discovery for me.

closed 2019-02-01 by marcosinigaglia