innoveit/react-native-ble-manager #301: Expo: Native module not found

mirhamzah, 2018-03-11:
Hi, I am new to react-native.

I'm trying to use this module in Expo (in IOS).
I am getting 'Native module not found. NativeEventEmitter.js:37:16' error. 

I have just copied this much code so far...

import {
  AppRegistry,
  StyleSheet,
  View,
  TouchableHighlight,
  NativeAppEventEmitter,
  NativeEventEmitter,
  NativeModules,
  Platform,
  PermissionsAndroid,
  ListView,
  ScrollView,
  AppState,
  Dimensions,
} from 'react-native';

import BleManager from 'react-native-ble-manager';

const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);

I have checked the file NativeEventEmitter.js and there is a special condition of Platform.OS === 'ios' which throws this error which makes me think it'd just work in Android ? I am not even sure if this module is supposed to work in Expo at all?

Any help will be appreciated.
Thanks,

marcosinigaglia, 2018-03-12:
Hi, I don't think Expo support external Native Modules.

VitamineC, 2018-03-15:
Expo only supports its native modules.

closed 2018-03-16 by marcosinigaglia

les-buchanan, 2019-02-26:
Thanks @marcosinigaglia.  You might reference that in the README to avoid future confusion.  I should have known it but it's been a while since I've worked with Expo.  Thanks again.