innoveit/react-native-ble-manager #302: Fatal Exception: RCTFatalException: Exception '*** Collection <__NSSetM: 0x101254fc0> was mutated while being enumerated.' was thrown while invoking connect on target BleManager with params ( "26D90E44-233E-689F-82EB-BBA5AC812C27", 5355 )

etayluz, 2018-03-12:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.8
- react-native v0.51.0
- iOS/Android v.12.2.2

### Expected behaviour

App should NOT crash when connecting to peripheral

### Actual behaviour

App crashes when tapping on peripheral to connect.

### Steps to reproduce

1. Scan for peripherals
2. Tap on peripheral to connect
3. Tap on peripheral to connect again and again

### Stack trace and console log
[image]

marcosinigaglia, 2018-03-16:
Hi, the module doesn't handle simultaneous commands

BaloMueller, 2018-09-07:
I have the same problem!

I'm absolutely sure that we do exactly one action at a time. But we have a whole lot of BLE Devices within reach of the smartphone.

This happens while doing the following actions:
- retrieveServices
- read
- connect
- disconnect

For us this is very urgent as we released a B2B Solution this week and a lot of users are using these Apps.

BaloMueller, 2018-09-10:
@marcosinigaglia I would like to fix this problem. Unfortunately I'm not sure what needs to be done. here.

I think the smartphone receives new peripherals while doing an action to another peripheral. So we should probably facilitate locks on some positions in code. But where?

marcosinigaglia, 2018-09-10:
Hi @BaloMueller , probably the device is stil scanning and retreaving the new peripherals, check in that part. I'm not good on object-c so I don't know how prevent correctly this exception.

BaloMueller, 2018-09-23:
For every operation I do to a peripheral, I first stop scanning and reactivate scanning afterwards. Feels like a hack, but solves the problem very easy.

closed 2019-01-09 by marcosinigaglia