innoveit/react-native-ble-manager #307: scan not returning intended peripherals

wke102, 2018-03-28:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.8
- react-native v0.54
- iOS/Android v5.0

I have a thermal printer, the device could be detected with normal android bluetooth scan, but when i am using the test function from the example, the device cannot be scanned.

Is this normal or should i pass any other things in?

marcosinigaglia, 2018-03-28:
Is the printer using the BLE or the standard Bluetooth?

wke102, 2018-03-28:
I'm not really sure, this library does not support standard Bluetooth devices? I also notice that it does not scan for idle devices, as in i need to do something on my other device in order for it to be scannable.

marcosinigaglia, 2018-04-03:
This library works only with BLE devices.

closed 2018-04-03 by marcosinigaglia