innoveit/react-native-ble-manager #311: Feature request for BleManagerDidReceiveString event listener or need help with BleManagerDidUpdateValueForCharacteristic.

unlabeled, 2018-04-11:
- react-native-ble-manager v6.2.9
- react-native v0.53.0
- iOS v11.3

### Here is what I'm trying to do

I'm using writeWithoutResponse command to send commands on my device connected to my HM10 or HM13 bluetooth card. Everything works great and my device changes volume for example when I'm sending `vol=4` to my card.
But sometimes I need to get a message from the device as an answer to my command. For example when I'm sending `ping` my device sends `echo` back.
* When I'm using an app which called `Serial` from App Store to send this command to my device it then receives the information from the card back and shows it on screen. So it shows `echo`.
I expected `BleManagerDidUpdateValueForCharacteristic` event listener to react on a message from my device in this case. But for some reason it never triggers. Other listeners are triggered when they should, `BleManagerDiscoverPeripheral`, `BleManagerStopScan`, `BleManagerConnectPeripheral` work fine in my app.

### Here are some questions

1. Should I expect `BleManagerDidUpdateValueForCharacteristic` work in my case? If so I guess the tag for my issue is "help needed" and I need some advice on how to make it work.
2. But I'm not sure that I should expect `BleManagerDidUpdateValueForCharacteristic` to work since I'm not really changing any characteristic on the card or on my device. So if I'm right about that then this issue is a "feature request".
I have checked the code for Serial app I talked before. Here it is https://github.com/hoiberg/HM10-BluetoothSerial-iOS. It's written with Swift language which I never saw before, but I guess I can understand what they did. They detect that my device can't receive `write` command and send `writeWithoutResponse` command. Then listener `serialDidReceiveString` is triggered in and they put the received string into text field, and scroll to the end of the text.
So can we have something like `BleManagerDidReceiveString` event listener here  as well?

Anyway, thanks for the authors, maintainers and collaborators for your great work. I really enjoy to use this library.

whitedogg13, 2018-04-12:
Hi, `BleManagerDidUpdateValueForCharacteristic` event will only be emitted after calling `startNotification` successfully, I think maybe it's the issue.

BTW, BLE itself won't send any user data back while `write` or `writeWithoutResponse`. To get user data from a BLE characteristic, you should either `read` (via `read` API) or register `notification` / `indication` (via `startNotification` API) from them.

unlabeled, 2018-04-16:
Yeah, thanks guys. The `BleManagerDidUpdateValueForCharacteristic` callback was not triggered since startNotification was missing.
I'm going to make PR for minor README update. I think somebody else would find this piece of information useful.

closed 2018-04-16 by unlabeled