innoveit/react-native-ble-manager #316: [Android] Value from a characteristic is received partially

ttdat89, 2018-04-17:
### Version
- react-native-ble-manager v6.2.9
- Android 7.0 24

### Expected behaviour
After writing value to a device, instead of receiving a whole value (in my case, it's an array of 100 bytes) like iOS, the value is split into 5 part, each part contains 20 bytes on Android.
Is there anyway to receive whole value as iOS did, or how to know when Android will end to read a long value? 

### Stack trace and console log
Android
[image]

iOS
[image]

marcosinigaglia, 2018-04-17:
Hi @ttdat89 , this is a thing that you have to handle in you app.

elailai94, 2018-04-17:
Hi @ttdat89, this is happening because there is packet fragmentation since the maximum MTU size on Android is 20 bytes. You will have to decide how you want to handle the reassembly of the packets when you receive them in your app, as @marcosinigaglia has pointed out earlier on. Here are some suggestions on how to handle the reassembly of the packets on the app's side:

1. If you're using Bluetooth 4, the maximum characteristic size is 512 bytes. Assuming your peripheral and Android phone supports this, you can request for a larger MTU size by calling the [requestMTU](https://github.com/innoveit/react-native-ble-manager#requestmtuperipheralid-mtu-android-only-api-21) function. In your case, this approach will eliminate the need to reassemble the packets as there is no packet fragmentation to begin with.

2. On the app side, you can allocate a buffer array of 100 bytes. Whenever you receive a packet of 20 bytes, you append it to the buffer array. Once the buffer array is filled, you can process and empty it.

ttdat89, 2018-04-18:
Thanks everyone

closed 2018-04-18 by ttdat89