innoveit/react-native-ble-manager #318: Disconnect cannot find peripheral after refresh

zacherkkila, 2018-04-26:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.9
- react-native v0.54.4
- Android v.8.1.0

### Expected behaviour

`disconnect` method also searches the actual list of connected devices from the GATT

### Actual behaviour

Note: I am not sure that this will be a problem in a real-world scenario, however, it is a bit of a pain for development. Also, apologies if I am missing something simple, haven't been using this for library for very long

After a refresh `getConnectedPeripherals` correctly returns the devices that are connected, however, disconnect returns `Peripheral not found`

Seems like it would be a good idea to also check `getBluetoothManager().getConnectedDevices(GATT)` for a matching peripheral device in the `disconnect` method and disconnect it if found

### Steps to reproduce

1. List connected peripherals (on click, disconnect)
2. Connect to a peripheral
3. Refresh the code
4. Click on the connected peripheral to disconnect (this is where it fails to find it)

dbarth, 2018-04-26:
I noticed that behavior and when looking at the code it became a bit more
obvious.

The behavior I ended up adopting is to automatically reconnect to the
device, and that re-populates the list of peripherals managed by the
library.

      BleManager.getConnectedPeripherals([SERVICE_UUID])
.then((peripheralsArray) => {
 if (peripheralsArray.length == 0)
   this.props.startScan();
 else {
   const peripheral = peripheralsArray[0];
   console.log("already connected to", peripheral);
   console.log("re-connecting...");
   BleManager.connect(peripheral.id)
     .then(() => {
this.whenConnected(peripheral);
     })
     .catch((error) => {
console.log(error);
     });
 }
})

Unless you do so, the library tries to find the peripheral to disconnect
from in it's own list and fails.

zacherkkila, 2018-04-30:
@dbarth Thanks for this workaround, I will likely do something similar for now. I still think that the library should work like this, maybe I will make a PR if I have time

closed 2019-02-01 by marcosinigaglia

maartenvandillen, 2022-07-05:
The problem (I think) I have is that after a React Native refresh the device is still connected but not in the internal list of the library. Scanning for it or disconnecting and reconnecting does not work because the device is already connected but not visible to the library. Only disabling and enabling Bluetooth fixes this. Do you have a work around for this?