innoveit/react-native-ble-manager #320: Cannot listen to event BleManagerDidUpdateValueForCharacteristic on IOS

ramjak, 2018-04-30:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.2.8
- react-native v0.50.4
- iOS v11.3
- XCode v9.3

### Actual behaviour

There is no error while using this package on Android Physical Device. But there is error while running this on IOS Physical Device.

### Steps to reproduce

1. Try to connect to device
2. Try to discover services
3. Try to listen to event 'BleManagerDidUpdateValueForCharacteristic'

### Stack trace and console log
The code stop on BleManager.m:line 97 when I tried to running with XCode:
`- (void)peripheral:(CBPeripheral *)peripheral didUpdateNotificationStateForCharacteristic:(CBCharacteristic *)characteristic error:(NSError *)error {
    if (error) {
        NSLog(@"Error in didUpdateNotificationStateForCharacteristic: %@", error);
        return;
    }
    
    NSString *key = [self keyForPeripheral: peripheral andCharacteristic:characteristic];
    
    if (characteristic.isNotifying) {
        NSLog(@"Notification began on %@", characteristic.UUID);
        RCTResponseSenderBlock notificationCallback = [notificationCallbacks objectForKey:key];
        notificationCallback(@[]);
        [notificationCallbacks removeObjectForKey:key];
    } else {
        NSLog(@"Notification ended on %@", characteristic.UUID);
        RCTResponseSenderBlock stopNotificationCallback = [stopNotificationCallbacks objectForKey:key];
        stopNotificationCallback(@[]); // line 97 << Thread 1: EXC_BAD_ACCESS (code=1, address=0x10)
        [stopNotificationCallbacks removeObjectForKey:key];
    }
}`

last XCode console message:
`2018-04-30 12:21:44.407731+0800 ProjectName[3655:413487] Notification began on 2A37
2018-04-30 12:21:44.524806+0800 ProjectName[3655:413487] Notification began on 2A5B
2018-04-30 12:21:44.585184+0800 ProjectName[3655:413487] Notification began on 2A37
warning: could not execute support code to read Objective-C class data in the process. This may reduce the quality of type information available.
`

And there is no error in chrome developer console. Thanks in advance.

ramjak, 2018-05-01:
This might be related to same issue in the parent project: https://github.com/don/cordova-plugin-ble-central/issues/389. Have try it, still the same.

marcosinigaglia, 2018-05-02:
Hi, can you try the fix they talk about in the issue?

ramjak, 2018-05-02:
I have try the fix, but the bug is still there :(

folofse, 2018-06-26:
I sometimes get the same problem and guess its because of the `BleManager.startNotification` getting fired twice in a row with the same service and charachteristics (even with `await`). I manage to recreate the problem with by adding two `BleManager.startNotification` after each other and removing the `await` expression. In your log its says `Notification began on 2A37` twice so it might be the problem. Try to find out why startNotification is being triggered twice. I also noticed that the views triggers event twice with React Navigation, so if you are using React Navigation `console.log` the `componentWillMount` function to see if the events get triggered twice.

ramjak, 2018-06-28:
Yes, I also get the same the same problem. But my BleManager is firing twice because there is a double-mounted component in my app. My solution is by directly changing native ios code in the BleManager.m:95 like this:
``` Objective-C
RCTResponseSenderBlock notificationCallback = [notificationCallbacks objectForKey:key];
if (notificationCallback != nil) {
  NSLog(@"Notification began on %@", characteristic.UUID);
  notificationCallback(@[]);
  [notificationCallbacks removeObjectForKey:key];
}
```
Not really a good solution because I have to change code the package code directly. Can I do PR for this? @marcosinigaglia

closed 2018-06-28 by ramjak

marcosinigaglia, 2018-06-29:
@ramjak sure,  make a PR.

ramjak, 2018-08-01:
#365