innoveit/react-native-ble-manager #321: nativeFlushQueueImmediate on write or writeWithoutResponse

sjgoalie, 2018-05-01:
- react-native-ble-manager v6.2.9
- react-native v0.54.2
- iOS/Android v8.1

Simplified code:

              encodedData = DataEncoder.encode(newData);
              var sendData = new Uint8Array(encodedData.length);
              console.log("in writeToCharacteristic - encodedData = " + encodedData.toString("hex"));
              for (var i = 0; i < encodedData.length; i++) { sendData[i] = encodedData[i]; }
              console.log("calling writeWithoutResponse - sendData = " + JSON.stringify(sendData));
              BleManager.writeWithoutResponse(this.currentDevice.id, characteristicInfo[1], characteristicInfo[0], sendData)
              .then((data) => {
                console.log('writeWithoutResponse - Wrote: ' + data);
              })
              .catch((error) => {
                console.log("writeWithoutResponse - error = " + error);
              });

Output:

 I/ReactNativeJS: in writeToCharacteristic - encodedData = 04000000
 I/ReactNativeJS: calling writeWithoutResponse - sendData = {"0":4,"1":0,"2":0,"3":0}
 E/ReactNativeJS: C++ exception in 'nativeFlushQueueImmediate'
                                                                   
                                                                   Malformed calls from JS: field sizes are different.
                                                                   
                                                                   [[34,39,39,39,28,39],[0,1,0,0,19,0],[[],[220],[221,100,1525209740592,false],[222,100,1525209740592,false],[223,100,1525209740597,false]],2079]

I tried with both writeWithoutResponse and write.

marcosinigaglia, 2018-05-02:
Try to use an Array instead Uint8Array

sjgoalie, 2018-05-02:
That changes the error to:

I/ReactNativeJS: writeToCharacteristic: writing to 50657263-6570-7420-5330-204d6f646520 = {"mode":"4"}
I/ReactNativeJS: in writeToCharacteristic - encodedData = 04000000
I/ReactNativeJS: calling writeWithoutResponse - sendData = [4,0,0,0]
E/unknown:ReactNative: Exception in native call
                                                                         java.lang.ClassCastException: com.facebook.react.bridge.ReadableNativeMap cannot be cast to java.lang.String
                                                                             at com.facebook.react.bridge.ReadableNativeArray.getString(ReadableNativeArray.java:136)
                                                                             at com.facebook.react.bridge.JavaMethodWrapper$5.extractArgument(JavaMethodWrapper.java:75)
                                                                             at com.facebook.react.bridge.JavaMethodWrapper$5.extractArgument(JavaMethodWrapper.java:71)
                                                                             at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:361)
                                                                             at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
                                                                             at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
                                                                             at android.os.Handler.handleCallback(Handler.java:790)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                             at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
                                                                             at android.os.Looper.loop(Looper.java:164)
                                                                             at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
                                                                             at java.lang.Thread.run(Thread.java:764)

marcosinigaglia, 2018-05-02:
Are characteristicInfo[1] and characteristicInfo[0] strings?

sjgoalie, 2018-05-02:
yes, sorry I meant to clean those up in the code paste, so it was more clear.   Those same strings work in a read.

sjgoalie, 2018-05-02:
got it, with yout last comment I went back over line by line, one of those characteristicInfo[] got stomped on somewhere above, my fault.  

Thanks for checking anyway.