innoveit/react-native-ble-manager #325: startNotification does not called

mikeex5, 2018-05-06:
- react-native-ble-manager v6.2.9
- react-native v0.55.2
- iOS/Android v4.4

I have problem with older Android devices. On Android 7.1 everything works fine, but on devices under version 6 startNotification() method is not called.

`BleManager.startNotification(deviceid, deviceService, deviceChar)
        .then(() => {
            // Success code
            console.log('Notification started');
        })
        .catch((error) => {
            // Failure code
            console.log(error);
        `})`

 In console log I don't see either the successful or the failed log.

alexbidiuk, 2018-05-11:
Subscribe

marcosinigaglia, 2018-05-18:
Hi, I saw that in some devices you don't have the feedback from the notification so the startNotification promise doesn't work but the BleManagerDidUpdateValueForCharacteristic event works.

gregogalante, 2018-06-04:
Get same problems.

kgaspar, 2018-12-07:
Hi did you guys managed to find a solution for this?
Got the same problem on android 5.1.1.

startNotification starts, but you don't get feedback about it every time, but the "BleManagerDidUpdateValueForCharacteristic" gets triggered. 

I observed the logcat on android and  the "setNotifyComplete()" logs get called, but the callback just doesn't fire up.

Tried on a 6.0 android and works fine. For some reason the callback doesn't fire.

gregogalante, 2018-12-07:
@kgaspar Yes, i don't know why but with an update of the minSdkVersion the application stops to request the  ACCESS_COARSE_LOCATION permission to the user.

I have forced it on application startup

```javascript
if (Platform.OS === 'android') {
      PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
        if (!result) {
          PermissionsAndroid.requestPermission(PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION).then((result) => {
            if (!result) {
              alert("The application can not work property without the requested permissions.")
            }
          })
        }
      })
    }
```

kgaspar, 2018-12-08:
@gregogalante 

Yeah they introduced that in Android 6.0. to protect the users privacy. As stated in [Android Developer](https://developer.android.com/about/versions/marshmallow/android-6.0-changes#behavior-hardware-id)

To scan the user needs to allow coarse location services.

But I think startNotification is not related to this. Since this is an 5.0 Android issue, and is related to not getting a notification started callback.

closed 2019-02-01 by marcosinigaglia