innoveit/react-native-ble-manager #326: incompatible types: boolean cannot be converted to int

madc, 2018-05-09:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.3.0

### Actual behaviour

The latst release breaks in our build with the following message:

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
:react-native-ble-manager:compileReleaseJavaWithJavac - is not incremental (e.g. outputs have changed, no previous execution, etc.).
/Users/vsts/agent/2.133.3/work/1/s/node_modules/react-native-ble-manager/android/src/main/java/it/innove/Peripheral.java:675: error: incompatible types: boolean cannot be converted to int
            int status = gatt.requestConnectionPriority(connectionPriority);
```

marcosinigaglia, 2018-05-09:
Thanks, try 6.3.1

madc, 2018-05-09:
Yep, back on track. That was really quick, thanks.

closed 2018-05-09 by madc