innoveit/react-native-ble-manager #328: NativeEventEmitter not adding listeners

zhan4139, 2018-05-14:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.3.1 & v6.3.0
- react-native v0.53.3
- iOS/Android v.11.3

### Expected behaviour

I was adding listener to discover peripherals in "componentDidMount" but did not working

### Actual behaviour

No peripherals was scanned and found

### Steps to reproduce

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

Here is my code:
```
const BleManagerModule = NativeModules.BleManager;
const bleManagerEmitter = new NativeEventEmitter(BleManagerModule);
...
 componentDidMount() {
    BleManager.start({ showAlert: false });
    this.handlerDiscover = bleManagerEmitter.addListener("BleManagerDiscoverPeripheral", () => {
    console.log("added listener"); // the arrow function was discoverPeripheral function
}
```

This is not adding any listener...

marcosinigaglia, 2018-05-14:
Hi, have you tried the example?

zhan4139, 2018-05-14:
Tried both on device and simulators, still not discovering. Xcode is giving some errors like ```nw_connection_get_connected_socket xxx Connection has no connected handler``` but I dont think it matters

zhan4139, 2018-05-15:
I think the problem got fixed by switching off bluetooth in ios setting and switching back on. But still wired why it didn't discovering devices. Maybe iOS remember it connected some devices then it stops discovering other devices even we re-build the project

closed 2018-05-16 by zhan4139