innoveit/react-native-ble-manager #331: Creates Bond but doesn't connect to device

vinipachecov, 2018-05-18:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.3.1
- react-native v0.55.3
- iOS/Android v.5.1.1

### Expected behaviour

It should connect to the device

### Actual behaviour
It fails to connect

### Steps to reproduce

1. Scan devices around 
2.  Create a bond with a specific device
3. Attempt to connect with this specific device

### Stack trace and console log

[image]

Here is the function which I'm calling the BleManager (following the steps described above).

[image]

The code that runs before this which leads to the unbounding of the device, this is just because I'm testing the connectivity performance of the lib and check if it will fulfill my needs. With native android code, using BluetoothSocket and BluetoothSocketServer I've been able to connect to it, but I'm having this problem to do the same with this lib.
The device I'm trying to connect is a bluetooth printer A5802:

https://www.jakartanotebook.com/zjiang-printer-resep-thermal-bluetooth-zj-5802-black

marcosinigaglia, 2018-05-18:
Hi @vinipachecov , try to debug the android native code.
PS: in ios createBond is not working.

vinipachecov, 2018-05-18:
Good to know this @marcosinigaglia ! However, I'm doing this on Android for now. Could explain how exatcly I should debug the native code? When you say the native code, you mean the java code of my project or the library code?

alvinhsu-kuracare, 2018-05-21:
Not working on Android while trying to connect to Polar H7.
- react-native-ble-manager v6.3.1
- react-native v0.55.4
- Android v.8.1.0

It always throws a generic "connection error" message.

rom-bar, 2018-05-24:
same here. i try to connect any printer bluetooth here .
using latest ble-manager
and android marshmellow.

closed 2019-06-18 by marcosinigaglia