innoveit/react-native-ble-manager #333: Update bluetooth firmware.

biks152207, 2018-05-25:
Hey guys i have been trying to update the firmware using this pacakge.But i am not sure how can i do it.I tried with writecharacteristic but i am not sure about the transaction id to do that.I have hex data of that firmware but i am very confused how we gonna inject those firmware hex data into ble device(my device is eld device, if you arenot sure about this go visit http://iosix.com/).Please suggest.

marcosinigaglia, 2018-05-28:
Hi @biks152207 , is not a standard procedure so you have check the device documentation.

vpfaulkner, 2018-10-04:
@biks152207 what did you end up doing for this? We need to do something similar and I can't find anything particular to this in the BLE specification. Thus, I'm guessing we just do so via writes to a dedicated characteristic?

MD3XTER, 2018-11-25:
@biks152207 take a look here https://github.com/Pilloxa/react-native-nordic-dfu

closed 2019-02-01 by marcosinigaglia

kumar-68150, 2024-09-18:
@andrei-zgirvaci  for STM-32 what lib need to be use?

andrei-zgirvaci, 2024-09-23:
@kumar-68150 not sure to be honest. Try to see how the firmware upgrade is done natively and then work backwards from there.