innoveit/react-native-ble-manager #335: [Android] removeBond not working

ducNgbk, 2018-05-29:
### Version
Tell us which versions you are using:

- react-native-ble-manager v6.2.8
- react-native v0.54.3
- Android v.6

### Expected behaviour

Hello,
I am doing a very simple function.
What I expect here is when I run below code, I'll get `removeBond success`, and my device will be deleted from Bluetooth setting of my phone.

### Actual behaviour

However, I got `removeBond success`, and my device was still there in Bluetooth Setting

I've tested on HTC desire 826 Android 6.0, Vivo 1610 Android 6.0.1

### Steps to reproduce

My code:
```
      console.log("Delete bond by code");
      await BleManager.removeBond(deviceInfo.addr)
        .then(() => {
          console.log("removeBond success");
        })
        .catch(() => {
          console.log("fail to remove the bond");
        });
```

marcosinigaglia, 2018-05-29:
Hi @ducNgbk , sorry but you have to debug the code by yourself to check if there is any problem.

ducNgbk, 2018-05-30:
Can you give me some clues, because I am not familiar with how RN and native platform communicate

closed 2019-02-01 by marcosinigaglia