innoveit/react-native-ble-manager #337: attempt to insert nil object from objects[2]' was thrown while invoking getConnectedPeripherals on target BleManager with params

anwarhamr, 2018-05-31:
I found existing closed [issue 134](https://github.com/innoveit/react-native-ble-manager/issues/134) that says this has been fixed since 3.2.1... 
### Version

- "react-native-ble-manager": "^6.2.9",
- "react-native": "0.52.1",
- iOS/Android I'm testing on both but only seen this error on iOS iPhone6

### Expected behaviour
promise to be resolved or rejected without exception

### Actual behaviour
This doesn't always happen.  I've only seen it after a crash and when I've restarted the app.

### Steps to reproduce

my code:
```
connect(peripheral) {
    this.log.info('BLE Connect requested', peripheral.id);
    return BleManager.getConnectedPeripherals([])
      .then((peripheralsArray) => {
        // Success code
        let idx = _.findIndex(peripheralsArray,{id:peripheral.id});
        this.log.info('BLE Connected peripherals: ',peripheralsArray,idx);
        if (idx >= 0){
          this.log.info('BLE WE ARE CONNECTED TO DEVICE ALREADY; DISCONNECTING', peripheral.id);
          return  BleManager.disconnect(peripheral.id);
        }
      })
      .then(() => BleManager.connect(peripheral.id))
      .then(()=> BleManager.getConnectedPeripherals([]))
      .then((devices) => {
        this.log.info('BLE Connected to', devices);
      })
      .then(() => BleManager.retrieveServices(peripheral.id))
      .then((peripheralInfo) => {
        this.log.info('BLE Service info:' ,peripheralInfo);
        return Promise.resolve(peripheralInfo);
      });
  }
```
As you can see in the stack trace I get the first log message but not the second found in the first .then().  I tested the _.findIndex() with possible values and it does not seem to be a problem.

### Stack trace and console log

```
09:11:27.616 debug-service.js:320 BLE Connect requested 90B8E3E4-6917-79AD-390E-A6D99EFFF75B

09:11:27.686 ExceptionsManager.js:73 Exception '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]' was thrown while invoking getConnectedPeripherals on target BleManager with params (
        (
    ),
    14803
)
callstack: (
	0   CoreFoundation                      0x00000001821d6da4 <redacted> + 252
	1   libobjc.A.dylib                     0x00000001813905ec objc_exception_throw + 56
	2   CoreFoundation                      0x000000018216f750 _CFArgv + 0
	3   CoreFoundation                      0x00000001820a80cc <redacted> + 352
	4   CoreFoundation                      0x00000001820a7f48 <redacted> + 64
	5   aura-test                           0x000000010108b050 dataToArrayBuffer + 268
	6   aura-test                           0x000000010108b558 -[CBPeripheral(com_megster_ble_extension) serviceAndCharacteristicInfo:] + 1180
	7   aura-test                           0x000000010108a490 -[CBPeripheral(com_megster_ble_extension) asDictionary] + 936
	8   aura-test                           0x000000010108e658 -[BleManager getConnectedPeripherals:callback:] + 908
	9   CoreFoundation                      0x00000001821de580 <redacted> + 144
	10  CoreFoundation                      0x00000001820bd748 <redacted> + 284
	11  CoreFoundation                      0x00000001820c256c <redacted> + 60
	12  aura-test                           0x0000000100eaedd4 -[RCTModuleMethod invokeWithBridge:module:arguments:] + 2064
	13  aura-test                           0x0000000100f53ad0 _ZN8facebook5reactL11invokeInnerEP9RCTBridgeP13RCTModuleDatajRKN5folly7dynamicE + 664
	14  aura-test                           0x0000000100f53660 _ZZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEiENK3$_0clEv + 132
	15  aura-test                           0x0000000100f535d0 ___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke + 28
	16  libdispatch.dylib                   0x00000001019d5260 _dispatch_call_block_and_release + 24
	17  libdispatch.dylib                   0x00000001019d5220 _dispatch_client_callout + 16
	18  libdispatch.dylib                   0x00000001019e3e80 _dispatch_queue_serial_drain + 768
	19  libdispatch.dylib                   0x00000001019d8730 _dispatch_queue_invoke + 328
	20  libdispatch.dylib                   0x00000001019e4dd8 _dispatch_root_queue_drain_deferred_wlh + 352
	21  libdispatch.dylib                   0x00000001019ebebc _dispatch_workloop_worker_thread + 676
	22  libsystem_pthread.dylib             0x0000000181dfbe70 _pthread_wqthread + 860
	23  libsystem_pthread.dylib             0x0000000181dfbb08 start_wqthread + 4
)
reactConsoleErrorHandler @ ExceptionsManager.js:73
console.error @ YellowBox.js:71
logToConsole @ RCTLog.js:48
logIfNoNativeHook @ RCTLog.js:31
__callFunction @ MessageQueue.js:353
(anonymous) @ MessageQueue.js:118
__guardSafe @ MessageQueue.js:316
callFunctionReturnFlushedQueue @ MessageQueue.js:117
t @ RNDebuggerWorker.js:1
```

marcosinigaglia, 2018-06-04:
Hi, in your example do you have a connected peripheral?

anwarhamr, 2018-06-06:
@marcosinigaglia I cannot say for sure but I am confident that where weren't any connected devices.

folofse, 2018-06-17:
I had almost the same problem, but it seems to be because of BleManager.retrieveServices, when I loged the recieved value from `[charachteristic value]` I got an empty NSData, so `dataToArrayBuffer` crashed and created the problem I had. I managed to fix it by adding this to line 198 in CBPeripheral+Extensions.m
```
if ([characteristic value] && [[characteristic value] length] > 0) {
          [characteristicDictionary setObject:dataToArrayBuffer([characteristic value]) forKey:@"value"];
      }
```
Maybe you have a similar problem and this can help you out.

anwarhamr, 2018-07-13:
@folofse 
should the file now look like this?
```
  if ((p & CBCharacteristicPropertyRead) != 0x0) {
   if ([characteristic value] && [[characteristic value] length] > 0) {
          [characteristicDictionary setObject:dataToArrayBuffer([characteristic value]) forKey:@"value"];
      }
  }
```
original file `CBPeripheral+Extensions.m`
```  
if ((p & CBCharacteristicPropertyRead) != 0x0) {
    [props addObject:@"Read"]; --line 198
  }
```

anwarhamr, 2018-07-13:
Is there a way I can attach the debugger?  I've tried to run from xcode and set a breakpoint but it doesn't to hit.

Sean-Brown, 2018-07-24:
@anwarhamr I've been struggling with debugging react-native on an iOS device, what I found works reliably most of the time is to attach to the packager that xcode launches. Once the app is installed on the device you can:

1. in your project's root directory, start the packager via `react-native start` (ensure existing packagers are stopped first!!)
2. open the app on the device
3. in Visual Studio Code run the "Attach to packager" debugging option
4. shake the device and choose the option to reload the javascript

Also you may want to consider using async/await instead of that promise chain (imo it makes the code easier to understand), e.g.:
```
async connect(peripheral) {
    this.log.info('BLE Connect requested', peripheral.id);
    try {
        const peripheralsArray = await BleManager.getConnectedPeripherals([]);
        let idx = _.findIndex(peripheralsArray,{id:peripheral.id});
        this.log.info('BLE Connected peripherals: ',peripheralsArray,idx);
        ...
    }
    catch (e) {
        // throw an exception, return an error value, do nothing, etc
    }
  }
```

Also I'm not sure why you'd want to disconnect from the device and then connect to it again -- if the goal is to connect to the device and it's already connected, why not just leave it connected?

anwarhamr, 2018-07-24:
@Sean-Brown thank you for the detailed example.  I'll try it this week.
Last week I did update to async/await.  My code is similar, good idea!
As for the disconnection.  I don't intentionally disconnect.  The device can get out of range or some other error occurs on the device side causing disconnection from central.
Thanks again.

NN01010101, 2018-09-20:
Not sure if my crash was related with this issue, but the solution published by @folofse fixed my crash. Thanks!

anwarhamr, 2018-09-20:
I've not seen this since adding @folofse solution above.  I think it is even in the codebase now.

closed 2018-09-20 by anwarhamr