innoveit/react-native-ble-manager #346: unfortunately stopped with android5.1

raguct25, 2018-06-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.3.3
- react-native v0.55.4
- iOS/Android v5.1

### Expected behaviour

i want developed bluetooth le device scanning project. i had used react-native-ble-manager . i want to work with android(API 19+) but here doesn't work with android5.1 device. android 6, android 7 is work well.

### Actual behaviour

If i press connect device button it  run with android 6, android 7 but if press in android 5.1 devices its unfortunately stopped popup throws and app closed what is problem

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2018-06-29:
Hi @raguct25 , you have to debug in your device, unfortunately the bluetooth can works differently in each implementation.

raguct25, 2018-06-29:
Hi...@marcosinigaglia 
**am using this code for connect to device:**

`test(peripheral) {
    if (peripheral){
      if (peripheral.connected){
        BleManager.disconnect(peripheral.id);
      }else{
        BleManager.connect(peripheral.id).then(() => {
          let peripherals = this.state.peripherals;
          let p = peripherals.get(peripheral.id);
          if (p) {
            p.connected = true;
            peripherals.set(peripheral.id, p);
            this.setState({peripherals});
          }
          console.log('Connected to ' + peripheral.id);

          setTimeout(() => {

            /* Test read current RSSI value
            BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
              console.log('Retrieved peripheral services', peripheralData);
              BleManager.readRSSI(peripheral.id).then((rssi) => {
                console.log('Retrieved actual RSSI value', rssi);
              });
            });*/

            // Test using bleno's pizza example
            // https://github.com/sandeepmistry/bleno/tree/master/examples/pizza
            BleManager.retrieveServices(peripheral.id).then((peripheralInfo) => {
              console.log(peripheralInfo);
              var service = '13333333-3333-3333-3333-333333333337';
              var bakeCharacteristic = '13333333-3333-3333-3333-333333330003';
              var crustCharacteristic = '13333333-3333-3333-3333-333333330001';

              setTimeout(() => {
                BleManager.startNotification(peripheral.id, service, bakeCharacteristic).then(() => {
                  console.log('Started notification on ' + peripheral.id);
                  setTimeout(() => {
                    BleManager.write(peripheral.id, service, crustCharacteristic, [0]).then(() => {
                      console.log('Writed NORMAL crust');
                      BleManager.write(peripheral.id, service, bakeCharacteristic, [1,95]).then(() => {
                        console.log('Writed 351 temperature, the pizza should be BAKED');
                        /*
                        var PizzaBakeResult = {
                          HALF_BAKED: 0,
                          BAKED:      1,
                          CRISPY:     2,
                          BURNT:      3,
                          ON_FIRE:    4
                        };*/
                      });
                    });

                  }, 500);
                }).catch((error) => {
                  console.log('Notification error', error);
                });
              }, 200);
            });

          }, 900);
        }).catch((error) => {
          console.log('Connection error', error);
        });
      }
    }
  }`

**am getting output:**
console.log('Connected to ' + peripheral.id);
 while suddenly i got unfortunately stopped popup

adam-bubela, 2018-07-01:
hi,
I have the same issue
Android 5.0.2
react-native-ble-manager@6.3.3
min SKD 19

1. run attached example
2. press 'Scan Bluetooth'
3. the App goes down -> 'Unfortunately, XYZ has stopped.'
4. Last logs:

`I/ReactNativeJS(30793): 'Got ble peripheral', { advertising:
I/ReactNativeJS(30793):    { localName: '[TV] Living room',
I/ReactNativeJS(30793):      isConnectable: true,
I/ReactNativeJS(30793):      manufacturerData:
I/ReactNativeJS(30793):       { bytes:
I/ReactNativeJS(30793):          [ 27,
I/ReactNativeJS(30793):            255,
I/ReactNativeJS(30793):            117,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            66,
I/ReactNativeJS(30793):            4,
I/ReactNativeJS(30793):            1,
I/ReactNativeJS(30793):            128,
I/ReactNativeJS(30793):            172,
I/ReactNativeJS(30793):            120,
I/ReactNativeJS(30793):            189,
I/ReactNativeJS(30793):            188,
I/ReactNativeJS(30793):            112,
I/ReactNativeJS(30793):            156,
I/ReactNativeJS(30793):            13,
I/ReactNativeJS(30793):            122,
I/ReactNativeJS(30793):            189,
I/ReactNativeJS(30793):            188,
I/ReactNativeJS(30793):            112,
I/ReactNativeJS(30793):            156,
I/ReactNativeJS(30793):            12,
I/ReactNativeJS(30793):            1,
I/ReactNativeJS(30793):            58,
I/ReactNativeJS(30793):            122,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0,
I/ReactNativeJS(30793):            0 ],
I/ReactNativeJS(30793):         data: 'G/91AEIEAYCseL28cJwNer28cJwMATp6AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=',
I/ReactNativeJS(30793):         CDVType: 'ArrayBuffer' } },
I/ReactNativeJS(30793):   rssi: -104,
I/ReactNativeJS(30793):   id: '78:BD:BC:70:9C:0D',
I/ReactNativeJS(30793):   name: '[TV] Living room' }
I/ReactNativeJS(30793): Scan is stopped
D/ReactNative(30793):  Catch to call normal connection
I/ReactNativeJS(30793): Connected to 78:BD:BC:70:9C:0D
I/ReactNativeJS(30793): Disconnected from 78:BD:BC:70:9C:0D`

danielkeithw, 2018-07-02:
This has started happening to me

react-native-ble-manager 6.2.8
Android 5.1.1

07-01 21:08:39.689 21637-21637/com.bbmobile E/AndroidRuntime: FATAL EXCEPTION: main
                                                              Process: com.bbmobile, PID: 21637
                                                              java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothGatt.discoverServices()' on a null object reference
                                                                  at it.innove.Peripheral$1.run(Peripheral.java:301)
                                                                  at android.os.Handler.handleCallback(Handler.java:739)
                                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                  at android.os.Looper.loop(Looper.java:135)
                                                                  at android.app.ActivityThread.main(ActivityThread.java:5422)
                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                  at java.lang.reflect.Method.invoke(Method.java:372)
                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:914)
                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:707)

I've had this working before....It just started suddenly

lyseiha, 2018-07-02:
Please help i have the same issues with my LG android version 5.1

raguct25, 2018-07-03:
**Solution is:** 
We had used this one
`"dependencies": {
    "react-native-ble-manager": "6.3.3"
  },`
this is not work with **android5.1**

**Use this:**

`"dependencies": {
    "react-native-ble-manager": "git://github.com/innoveit/react-native-ble-manager.git#master"
  },`

i used this one the problem was fixed

raguct25, 2018-07-03:
@adam-bubela  and @danielkeithw plz check it issue was fixed

closed 2018-07-03 by raguct25

marcosinigaglia, 2018-07-03:
Hi, I made a new release with the fix.

danielkeithw, 2018-07-04:
Thanks @marcosinigaglia this does stop by crash. I appreciate it :)

I'm going to work on some retry logic in the JS code because its an awkward user interaction to have to connect twice.

Thanks Again!