innoveit/react-native-ble-manager #354: Is it possible to pair ble device programmatically without the user entering pin??

Hyeongwon, 2018-07-13:
Hi, Guys

my goal is for the app to automatically input the password when pairing with the bluetooth device without asking user to input the PIN.

Thanks!

wesdewitte, 2018-10-11:
@Hyeongwon Did you fix it? How did you do it then?

Hyeongwon, 2018-10-12:
> @Hyeongwon Did you fix it? How did you do it then?

no I didn't

erdouzun, 2019-04-24:
Hello,

Is there any update for this capability?

mazenchami, 2019-06-18:
any update on this?

marcosinigaglia, 2019-06-19:
Is possible but I don't have a device to test, a PR is welcome.

https://stackoverflow.com/questions/35519321/android-bluetooth-pairing-without-user-enter-pin-and-confirmation-using-android

siddaiahm, 2020-10-05:
Hey, I have made a PR for this changes, I tested and it is working in android

closed 2020-10-05 by marcosinigaglia

maartenvandillen, 2022-02-14:
@siddaiahm this used to work perfectly for me, but not anymore. I'm not sure if the update of the firmware is the cause or maybe an update of my phone (Samsung). Is it still working for you?

siddaiahm, 2022-02-21:
> @siddaiahm this used to work perfectly for me, but not anymore. I'm not sure if the update of the firmware is the cause or maybe an update of my phone (Samsung). Is it still working for you?

Yes still working as expected for me (v7.4.2).  I did not check in the other updated versions.