innoveit/react-native-ble-manager #356: "serviceUUIDs" on iOS, but "serviceUuids" on Android

snarpch, 2018-07-16:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.4.0
- react-native v0.56.0
- iOS/Android v.?

### Expected behaviour

When receiving "BleManagerDiscoverPeripheral" events, the JavaScript objects for the peripherals contain advertisement data. If the device advertises serviceUUIDs, the arrays in the object are named differently. On iOS, it is called "serviceUUIDs", and on Android it is called "serviceUuids" (note the case-sensitive difference...).

To make cross-platform development easier, it would be better if both of these would be written the same way...

### Steps to reproduce

1. Register for Peripheral-Discovered events: bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', handleDiscoverPeripheral);

2. start scanning: BleManager.scan([], 10, true)...

3. Note differences on iOS and Android in handleDiscoverPeripheral(peripheral): console.log(peripheral.advertising);

marcosinigaglia, 2018-07-24:
Hi, try to use the latest release.

closed 2018-07-24 by marcosinigaglia

savitaraghuvanshi, 2021-06-07:
does this library works cross platform ? transfer msg between ios to android and vice versa ?